nanoframework/Home #385: Math.Round(float) hangs the board

steffalk, 2018-08-22:
## Details about Problem

Visual Studio 2017 15.8.1
Windows 10 1803 17134.228

The problem did not occur some weeks to months ago, but unfortunately I cannot say the exact version numbers.

## Detailed repro steps so we can see the same problem

1. Create a solution with the following code:

`
using System;

namespace NFAppTestMathHang
{
    public class Program
    {
        public static void Main()
        {
            Console.WriteLine("Hello world!");

            float x;

            x = Math.Round(5.5f); // <-- hangs here to the point that the board must be re-flashed!

            Console.WriteLine(x.ToString()); // <-- never arrives here
        }
    }
}
`

2. Run the code. The board hangs at Math.Round.

piwi1263, 2018-08-22:
Hi guys,

this is somewhat hard to replicate ...

[image]

Using VS 2017 15.7.6 and extension 0.5.0.24.

steffalk, 2018-08-23:
This is funny. It happens on both my ND3 and ND3 WiFi, totally reproducible.

steffalk, 2018-08-23:
Can I check anything?

josesimoes, 2018-08-24:
Confirmed that this is being caused by the compiler optimizations.
Reworking the math calls seems to be the best option to fix this (and other similar problems with calls to other math methods).

closed 2018-08-24 by josesimoes