nanoframework/Home #389: Unable to start debug session on ESP32 when using SerialCommunication

josesimoes, 2018-08-29:
## Details about Problem

nanoFramework area Visual Studio extension | Hardware/target board

VS version (if appropriate): 
OS version (i.e. win10 v1607 (14393.321)):

Worked before? yes

## Detailed repro steps so we can see the same problem

1. Can't start debug of SerialCommunication sample app

## Other suggested things
- This seemed to start after adding back the GenericPort_Write in commit [e39705b](https://github.com/nanoframework/nf-interpreter/commit/e39705b0b2abe4b0e22fbb754b2c5aea99d375a0) and...
- ... adding the events for watch char and data received.
- The call to `ets_printf` was commented before, now it's not
- This code also exists on developer-networking _but_ I'm not sure if it was tested using serial

CC
@AdrianSoundy 
@MatthiasJentsch

AdrianSoundy, 2018-08-29:
This is because the VS extension and serial/generic port are using the same serial connection. VS extension copes with it if its not too much extra data but as soon as you try something larger you have problems with debugger. 

This needs to be commented out for now until we can work out a better solution.

What we need is to not output if the debugger is connected.

Also we could optional send the data to the ESP32 apptrace so it can be picked up by the Jtag/openocd serial  output

josesimoes, 2018-08-29:
> Also we could optional send the data to the ESP32 apptrace so it can be picked up by the Jtag/openocd serial output

I like this one much better!

closed 2018-09-09 by MatthiasJentsch

doingnz, 2018-09-09:
Another approach to consider is to more easily direct the debug trace to another UART. 
I have not had the time to determine how ets_print() is setup to use a specific UART.
In my testing currently, I have used the functionality in WP to move the nanoFramework to UART_1.
Then ets_print() uses the default UART_0. Obviously you need to attached a USB-to-Serial cable to the UART_1 pins for the ESP32. 

this all "works" in that I can have Debug serial out one port (UART_0) and have VS2017 on the other UART. What is irritating is that the flashing needs to happen through UART_0. You need to go through a song and dance routine each time you flash new FW where you remember to close the terminal app monitoring UART_0 before you flash the firmware. It would be easier if WP remained on UART_0 and ets_print() was directed to UART_1. Should be trivial to do and ideally should follow a process similar to what is used in the WP file to select UART_0/1/2.