nanoframework/Home #392: Math.Truncate should return the integral part

MatthiasJentsch, 2018-09-01:
## Details about Problem

nanoFramework area: nanoCLR

Worked before? No

## Detailed repro steps so we can see the same problem

Math.Truncate should return the integral part. But I've found
```float d = stack.Arg0().NumericByRefConst().r4;
float res = 0.0;
float retVal = modff(d, &res); 

stack.SetResult_R4( retVal );
```
in `Library_corlib_native_System_Math::Truncate___STATIC__R4__R4`
This code returns the fractional part.

NemesisXB, 2018-09-03:
I will fix this.

josesimoes, 2018-09-03:
This is because of a recent code change replacing the original code with the calls to the CRT standard lib functions. If you check the previous versions you'll find there the code bit that was handling this nuance specific of .NET.

josesimoes, 2018-09-03:
This is because of a recent code change replacing the original code with the calls to the CRT standard lib functions. The args in the call are reversed if I remember correctly... 😊

closed 2018-09-03 by josesimoes