nanoframework/Home #405: Random.Next returns negative numbers

MatthiasJentsch, 2018-09-04:
## Details about Problem

nanoFramework area: nanoCLR preview 791 / nanoFramework.CoreLibrary preview 62

Worked before? No

## Detailed repro steps so we can see the same problem

The Random.Next returns not only values >= 0. I think that's caused by the nanoFramework.CoreLibrary Random.Next implementation. There is the uint32 return value from the unmanged part returned as int32

In nanoFramework.CoreLibrary:
```
/// <summary>
/// Returns a nonnegative random number.
/// </summary>
/// <returns>A 32-bit signed integer greater than or equal to zero and
/// less than MaxValue.</returns>
[MethodImpl(MethodImplOptions.InternalCall)]
public virtual extern int Next();
```

But in nanoCLR the return type is **uint32_t**

josesimoes, 2018-09-05:
You've spotted an interesting situation! 😄 
The [API doc](https://docs.microsoft.com/en-us/dotnet/api/system.random.next?view=netframework-4.7.2) states that's an `Int32` with max of `Int32.MaxValue`.

The lower layer functions (on all targets) return an `uint32_t` which is happily loaded on the return var with calls to `stack.SetResult_I4( ... );`. If the random value happens to have it's MSB set it will turn into a negative value.

Without putting too much thinking into this, it seems that we can 1) cast, 2) request a new one until it's in range, 3) change the code on the lower layer to keep it "in range" or none of the previous... 😉

doingnz, 2018-09-05:
I would vote for 3 and suggest the following pseudo-code (value a place holder for the existing variable that is uint32_t)
````
if (value > (uint32_t)Int32.MaxValue) {value &= (uint32_t)Int32.MaxValue;} 
```
Ensures value is less than or equal to Int32.MaxValue, and the casts ensure no compiler warnings about comparison of signed/unsigned variables.

closed 2018-09-05 by MatthiasJentsch