nanoframework/Home #406: double.ToString() and float.ToString() not working as described

MatthiasJentsch, 2018-09-04:
## Details about Problem

nanoFramework area: nanoCLR preview 791 / nanoFramework.CoreLibrary preview 62

Worked before? No

## Detailed repro steps so we can see the same problem

```(47.0).ToString()``` returns "47.000000000000000" and ```(11.0f).ToString()``` returns "11.000000000". But in lib-CoreLibrary\source\System\Number.cs for format "G" is stated: "Trailing zeros after the decimal point are removed, and the resulting string contains a decimal point only if required."

StephenCardinale, 2018-11-17:
float.ToString() and double.ToString() exhibiting odd behavior and extra zero (0) padding applied. See attached example: [Program.zip](https://github.com/nanoframework/Home/files/2592366/Program.zip)

Here is the output that I get using various format specifiers:

Float - format specifier: 'N' of '1234.0123456789' is 1,234.0100
Float - format specifier: 'N0' of '1234.0123456789' is 1,234
Float - format specifier: 'N1' of '1234.0123456789' is 1,234.00
Float - format specifier: 'N2' of '1234.0123456789' is 1,234.0100
Float - format specifier: 'N3' of '1234.0123456789' is 1,234.012000
Float - format specifier: 'N4' of '1234.0123456789' is 1,234.01230000
Float - format specifier: 'N5' of '1234.0123456789' is 1,234.0123300000
Float - format specifier: 'N6' of '1234.0123456789' is 1,234.012329000000
Float - format specifier: 'N7' of '1234.0123456789' is 1,234.01232910000000
Float - format specifier: 'N8' of '1234.0123456789' is 1,234.0123291000000000
Float - format specifier: 'N9' of '1234.0123456789' is 1,234.012329101000000000

Double - format specifier: 'N' of '1234.0123456789' is 1,234.0100
Double - format specifier: 'N0 of '1234.0123456789' is 1,234
Double - format specifier: 'N1' of '1234.0123456789' is 1,234.00
Double - format specifier: 'N2' of '1234.0123456789' is 1,234.0100
Double - format specifier: 'N3' of '1234.0123456789' is 1,234.012000
Double - format specifier: 'N4' of '1234.0123456789' is 1,234.01230000
Double - format specifier: 'N5' of '1234.0123456789' is 1,234.0123500000
Double - format specifier: 'N6' of '1234.0123456789' is 1,234.012346000000
Double - format specifier: 'N7' of '1234.0123456789' is 1,234.01234570000000
Double - format specifier: 'N8' of '1234.0123456789' is 1,234.0123456700000000
Double - format specifier: 'N9' of '1234.0123456789' is 1,234.012345678000000000

Float - format specifier: 'F' of '1234.0123456789' is 1234.0100
Float - format specifier: 'F0 of '1234.0123456789' is 1234
Float - format specifier: 'F1' of '1234.0123456789' is 1234.00
Float - format specifier: 'F2' of '1234.0123456789' is 1234.0100
Float - format specifier: 'F3' of '1234.0123456789' is 1234.012000
Float - format specifier: 'F4' of '1234.0123456789' is 1234.01230000
Float - format specifier: 'F5' of '1234.0123456789' is 1234.0123300000
Float - format specifier: 'F6' of '1234.0123456789' is 1234.012329000000
Float - format specifier: 'F7' of '1234.0123456789' is 1234.01232910000000
Float - format specifier: 'F8' of '1234.0123456789' is 1234.0123291000000000
Float - format specifier: 'F9' of '1234.0123456789' is 1234.012329101000000000

Double - format specifier: 'F' of '1234.0123456789' is 1234.0100
Double - format specifier: 'F0 of '1234.0123456789' is 1234
Double - format specifier: 'F1' of '1234.0123456789' is 1234.00
Double - format specifier: 'F2' of '1234.0123456789' is 1234.0100
Double - format specifier: 'F3' of '1234.0123456789' is 1234.012000
Double - format specifier: 'F4' of '1234.0123456789' is 1234.01230000
Double - format specifier: 'F5' of '1234.0123456789' is 1234.0123500000
Double - format specifier: 'F6' of '1234.0123456789' is 1234.012346000000
Double - format specifier: 'F7' of '1234.0123456789' is 1234.01234570000000
Double - format specifier: 'F8' of '1234.0123456789' is 1234.0123456700000000
Double - format specifier: 'F9' of '1234.0123456789' is 1234.012345678000000000

Float - format specifier: 'G' of '1234.0123456789' is 1234.012329101000000000
Float - format specifier: 'G0 of '1234.0123456789' is 1234.012329101000000000
Float - format specifier: 'G1' of '1234.0123456789' is 1234.012329101000000000
Float - format specifier: 'G2' of '1234.0123456789' is 1234.0100
Float - format specifier: 'G3' of '1234.0123456789' is 1234.012000
Float - format specifier: 'G4' of '1234.0123456789' is 1234.01230000
Float - format specifier: 'G5' of '1234.0123456789' is 1234.0123300000
Float - format specifier: 'G6' of '1234.0123456789' is 1234.012329000000
Float - format specifier: 'G7' of '1234.0123456789' is 1234.01232910000000
Float - format specifier: 'G8' of '1234.0123456789' is 1234.0123291000000000
Float - format specifier: 'G9' of '1234.0123456789' is 1234.012329101000000000

Double - format specifier: 'G' of '1234.0123456789' is 1234.012345678899919000000000000000
Double - format specifier: 'G0 of '1234.0123456789' is 1234.012345678899919000000000000000
Double - format specifier: 'G1' of '1234.0123456789' is 1234.012345678899919000000000000000
Double - format specifier: 'G2' of '1234.0123456789' is 1234.0100
Double - format specifier: 'G3' of '1234.0123456789' is 1234.012000
Double - format specifier: 'G4' of '1234.0123456789' is 1234.01230000
Double - format specifier: 'G5' of '1234.0123456789' is 1234.0123500000
Double - format specifier: 'G6' of '1234.0123456789' is 1234.012346000000
Double - format specifier: 'G7' of '1234.0123456789' is 1234.01234570000000
Double - format specifier: 'G8' of '1234.0123456789' is 1234.0123456700000000
Double - format specifier: 'G9' of '1234.0123456789' is 1234.012345678000000000
`

josesimoes, 2018-11-19:
@MatthiasJentsch could you please look into this? Could it be a regression or something that hasn't being tested before?

MatthiasJentsch, 2018-11-19:
Sorry. I've tested only these cases:

`
Convert.ToDouble("4711E+81")
Convert.ToDouble("-815E-47")
4E+81.ToString()
(-8E-47).ToString()
3E7.ToString("F0")
`

I'll test the described behavior and will try to find a fix for this.

closed 2018-11-19 by josesimoes