nanoframework/Home #408: Convert::NativeToInt64 doesn't support base 2 and base 8 conversions

MatthiasJentsch, 2018-09-10:
## Details about Problem

nanoFramework area: nanoCLR preview 1000

Worked before? No

## Detailed repro steps so we can see the same problem
```Convert.ToByte("11001010", 2)``` and ```Convert.ToByte("123", 8)``` results in a ```NotImplementedException``` because in method ```HRESULT Library_corlib_native_System_Convert::NativeToInt64___STATIC__I8__STRING__BOOLEAN__I8__I8__I4( CLR_RT_StackFrame& stack )``` are only base 10 and base 16 conversions supported.

josesimoes, 2018-09-10:
@MatthiasJentsch this is by design, not a bug.
There is the build option `SUPPORT_ANY_BASE_CONVERSION` see [here](https://github.com/nanoframework/Home/blob/master/docs/building/cmake-tools-cmake-variants.md#content-explained) to enable support for converting from _any_ base. When that is OFF the supported conversions are base 10 and 16.

The rational behind this? The vast majority of use cases are for those 2 bases. Adding support for any base conversion adds significant flash and ram which is not good for targets with low resources.

Considering that this is available as option and that for most reference targets that is enabled, I'm closing this with a won't fix label.

closed 2018-09-10 by josesimoes

doingnz, 2018-09-10:
Need to remember this option when doing unit testing. Maybe the unit testing can be setup with #if/endif using SUPPORT_ANY_BASE_CONVERSION to include/exclude the additional bases in the build and unit tests. Ideally when running unit tests, there is a platform that can run with this enabled and therefore unit tested. 

When building/running the unit tests, the tests for the skipped bases should be flagged as having been skipped. forces the tester to make a conscious decision about checking or not checking the tests for a given platform.

NemesisXB, 2018-09-10:
How much more flash and RAM will we lose if using something like this:

[baselibc-strtoumax](https://github.com/PetteriAimonen/Baselibc/blob/master/src/strntoumax.c)
The (base == 8) case is missing but can be added.

Worth noting that we use baselibc in our projects. In one of our bigger projects, we saved close to 20kB flash (I did not look at the RAM savings or losses)

josesimoes, 2018-09-10:
Interesting... We should do a comparison to test all the different possibilities: with, without, just 2&810&16, all, etc

josesimoes, 2018-09-10:
That's great. I've opened a note on the Engine project to have a reminder for this. If you get there first, please promote that to task.