nanoframework/Home #410: Error loading nanoframework extension right after install

anteph, 2018-09-11:
Hi,

After installing the nanoframework extension from the visual studio Tools menu, it says that is is not possible to load it and that restarting visual studio may help. Unfortunately, neither restarting visual studio or the computer helps.

It also recommends checking an activity log file under the C:\user\AppData\Roaming\Microsoft\VisualStudio\15.0_xxxxx folder.

I've taken a look at that file (it's a .xml) and it seems to be missing the Microsoft.VisualStudio.Threading dll. I would try to install this dependency manually but since this is an extension and the error is not associated with any solution, I'm not sure if this is possible.

I've tried also in another computer after re-installing visual studio from scratch, but I get the same generic error when opening visual studio. Nonetheless, in that machine, the activity log indicates different missing files, so I'm focusing in my original attempt, since the solution is probably similar.

Sorry for not posting screenshots but my OS is in Portuguese.

Visual Studio version: Community Edition 2017
OS version : Windows 8.1

Thanks,
Nuno Santos

josesimoes, 2018-09-11:
What is the version of VS that you are using?

anteph, 2018-09-11:
I was on version 15.0.x (don't recall the exact value) but I've finished updating to 15.8.3 just a few minutes ago and it is no longer showing the loading error when starting VS.

I'll see if I can complete the rest of the procedure and get started, and get back with some more info.

Thanks,
Nuno Santos

josesimoes, 2018-09-11:
Being on the bleeding edge is a requirement around here! 🤣 😉 

The reported issue is caused by this version mismatch, for sure.

closed 2018-09-11 by josesimoes

anteph, 2018-09-11:
lesson learned : )

I seem to be able to create a new nanoframework project and the device explorer windw is now showing. Nonetheless, even after uploading the ESP32 firmware, no device shows in the list and all the buttons are grayed out, except for the "show internal errors" one.

I'll do some more digging to see if I can solve the problem. 

Thank you very much for your help!

josesimoes, 2018-09-11:
Make sure you've unplug and plug back your ESP32 after flashing it.

Also suggest that if you keep experiencing this , ask in the discord esp32 support channel. You'll get a quick support there.