nanoframework/Home #411: GpioController and GpioPin providers

RoySalisbury, 2018-09-12:
Part of the design in the "Windows.Devices.*" UPW API was the ability to add your own device "controllers" such as a GpioController, via some provider interfaces.

These interfaces were IGpioControllerProvider, IGpioProvider, and IGpioPinProvider. For an example of these interfaces and the pattern, see the following:

https://github.com/ms-iot/BusProviders/blob/develop/SimulatedProvider/SimulatedProvider/GpioProvider.cs

https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/lightningproviders

However, due to code space and the nature of the devices (small single purpose boards) this was not implemented.  I can completely understand this.  And honestly, the design in UWP is very confusing.  The current UWP classes for GpioController and GpioPin have interfaces on them, but MS marked them internal use only. I would suspect for the type conversion from the provider model.

I am proposing that the IGpioController and IGpioPin interfaces be made public in the nF so that there is a consistent interface for controllers and pins that may be implemented within the community.  One of the biggest reasons is code separation.  A device driver should not have to be responsible for instantiating the controller or pins that may be needed. Another is that not everyone likes to use the main boards GPIO's for non-interrupt uses.

For example. The HD44780 LCD displays usually take between 7 and 11 gpio's for the display control.  Some implementations are hard coded to use the native Gpio controller and either specific pins, or you specify the pins during construction/initialization.  This makes the display now responsible for the controller/pins and locks in the driver to specific hardware.  However, by separating the responsibility of the controller/pin creation from the display driver, I would just pass in the already created IGpioPin's.  These pins could come from anywhere.  2 from the default controller, 4 from an MCP23008 gpio expander, and the rest from some other chip.

I'm sure this all makes perfect sense to anyone that has been developing for more that 1 day.  Its just all standard framework designs.

I will be willing to implement all these changes (very minimal that they are) for the GpioController and I suggest that the same design be followed for the SPI, I2C, and ADC controllers. 

For the Gpio controller this will consist of 2 new interfaces (IGpioController and IGpioPin). The current GpioController and GpioPin classes will implement these interfaces (they sort of actually already do), and the GpioController will either need 3 new proxy methods to conform to the interface, or the current signatures changed from GpioPin to IGpioPin. I would suggest the 3 new proxy calls so no contracts are broken.  If code space is an issue, then changing the signatures will work as well. Its a very minor change and technically is still compatible code.

RoySalisbury, 2018-09-12:
I got as far as cloning the repo, creating a branch, creating a test project for a GpioController (just an led blinking) so I had a baseline using the existing code.

Then I added the necessary interfaces, modified the existing classes, added all the XML comments, and changed from the NuGet reference to my local project reference.  Compiles fine.  But I cant deploy to test anything.  Guess its not just a straight forward "update the assembly" type of thing.

So, I'm at a stopping point for the moment.  I won't be able to get back to this until the weekend.  I'm 99.99% positive its all good.  But until I can test it locally with the default Gpio controller to make sure, and against my MCP23008 gpio controller code, I'm not sure how to proceed.  

I could commit what I have to my branch (develop-gpio-interfaces), but that seems to be bad form (don't commit a broken build). 

Perhaps I'll move onto the "nf-Community-Contributions" to get the MCP23008 Gpio controller ready for submission using the interfaces (at least up to a point .. won't be able to test that either).

Any guidance on how to proceed would be helpful.

josesimoes, 2018-09-12:
@RoySalisbury this makes perfect sense to me, as I've already wrote on the Discord channel. Let's wait for input from others.

As for the issues with deployment, I'll be guiding you through those on the support channel on Discord.

RoySalisbury, 2018-09-14:
For the life of me I can't figure out all this git/clone/branch stuff to submit a PR, so if anyone wants to take my code and incorporate it into the nF, your more than welcome.   

The 4 files are on my dropbox here:

https://1drv.ms/u/s!Aq5b3zEUkJG5gp8orwNRZuiGFmFcIQ

Shouldn't be this hard.  :(

RoySalisbury, 2018-09-14:
OK.. I "think" I may have figured this out.  Its not like all the other projects that I have worked with when submitting PR's.  

Unlike the other projects that I have dealt with, you are required to FORK the repository and then submit a PR based on that fork's changes.  I will follow this model and see if I can get the changes uploaded.

RoySalisbury, 2018-09-15:
OK.. Did my best to get the code included/updated and submitted the 2 PR's needed.

josesimoes, 2018-09-15:
> Unlike the other projects that I have dealt with, you are required to FORK the repository and then submit a PR based on that fork's changes. I will follow this model and see if I can get the changes uploaded.

We are following here the same workflow and GP of 95% of Open Source projects out there... the model FORK -> CLONE LOCAL -> CHANGES -> PR to UPSTREAM is the absolute common workflow.

josesimoes, 2018-09-18:
Feature implemented made available.

closed 2018-09-18 by josesimoes