nanoframework/Home #418: Unable to deploy code on STM32F401RE board

sharmavishnu, 2018-10-06:
## Details about Problem
Built nanoFramework on 5-Oct-18,(@1032 commits,#886) I'm using nanoFramework extension version 0.5.0.49.  After flashing the nanoCLR.hex and nanoBooter.hex, when I now connect the board (STM32F401RE) I can perform ping, erase and reboot operations in the device explorer, however, when I click on "Device Capabilities", it fails with the error:
**Error retrieving device information from NANO_30663595323 @ COM4. Please reconnect device.

VS version: Enterprise 2017, V 15.8.2

OS version: Windows 10 Pro, V 1803

This used to work before. The last build I made was on 17-Sep-18. That used to work

Detailed steps to reproduce

1. Create a build for STM32F401RE. Attached variants file will provide information on the settings
2. Flash the hex files using ST-Link. Connect the board via second USB cable for COM port
3. Open Visual Studio and you should see a Nano device in device explorer
4. Click on Ping and you should see a response.
5. Click on reboot, the message in the output panel shows successful reboot.
6. Click on "Device Capabilities" and you get the error as given below:
2>Getting things ready to deploy assemblies to nanoFramework device: NANO_30663595323 @ COM4.
2>nanoDevice is initialized
2>Couldn't find any native assemblies deployed in NANO_30663595323 @ COM4! If the situation persists reboot the device.

Rebooting the device does not help.

sharmavishnu, 2018-10-07:
If it helps, found out that the above issue happens when I try to build MinRelSize. When I build the "Debug"  variant, the issue disappeared.

josesimoes, 2018-10-07:
From the above this is most likely a misaligned block storage declaration with the linker file.

closed 2018-10-07 by josesimoes