nanoframework/Home #419: Deploy failed on ESP32: Couldn't find a valid native assembly required by mscorlib v1.0.0.0

ruslankiskinov, 2018-10-07:
I'm trying to deploy an empty (blank) C# project to the **ESP32 WROOM-32** board.
I successfully flashed the board with the **EspFirmwareFlasher**. 

Visual studio is detecting the board and ping says: 
**"Silicon Labs CP210x USB to UART Bridge (COM3) @ COM3 is active running nanoCLR"**

Before that I was able to deploy applications with Arduino IDE with no problem and they are working.

When I try to deploy a completely blank solution created from the nanoFramework VS template (also build with no problems) I receive an error during deploy:
**"Couldn't find a valid native assembly required by mscorlib v1.0.0.0"**

*Here is the output from nanoFramework console:*
`07 03:56:43Z [NanoDevices: open device \\?\USB#VID_10C4&PID_EA60#0001#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]`

*Here is the output of the Build console:*
```
1>------ Build started: Project: NanoTest, Configuration: Debug Any CPU ------
1>NanoTest -> d:\Dropbox\Electronics\ESP32\NanoTest\NanoTest\bin\Debug\NanoTest.exe
2>------ Deploy started: Project: NanoTest, Configuration: Debug Any CPU ------
2>Getting things ready to deploy assemblies to nanoFramework device: Silicon Labs CP210x USB to UART Bridge (COM3) @ COM3.
2>nanoDevice is initialized
2>Deploy failed.
2>***************************************
2>Couldn't find a valid native assembly required by mscorlib v1.0.0.0, checksum 0x7F67F424.
2>***************************************
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========
```

VS version (if appropriate): 2017 Community 15.8.5
OS version (i.e. win10 v1607 (14393.321)): Windows 10 Version 10.0.17134.285

[image]

ruslankiskinov, 2018-10-07:
Forgot to put the booting log from the ESP32 if it can be useful...

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5808
ho 0 tail 12 room 4
load:0x40078000,len:9020
load:0x40080000,len:6120
entry 0x40080334
[0;32mI (29) boot: ESP-IDF v3.1 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 14:03:14[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (38) boot: SPI Mode       : DIO[0m
[0;32mI (42) boot: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (79) boot:  3 deploy           Unknown data     01 84 00110000 00200000[0m
[0;32mI (87) boot: End of partition table[0m
[0;32mI (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x219a8 (137640) map[0m
[0;32mI (148) esp_image: segment 1: paddr=0x000319d0 vaddr=0x3ffb0000 size=0x03684 ( 13956) load[0m
[0;32mI (154) esp_image: segment 2: paddr=0x0003505c vaddr=0x3ffb3684 size=0x00000 (     0) load[0m
[0;32mI (155) esp_image: segment 3: paddr=0x00035064 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (164) esp_image: segment 4: paddr=0x0003546c vaddr=0x40080400 size=0x0aba4 ( 43940) load[0m
[0;32mI (191) esp_image: segment 5: paddr=0x00040018 vaddr=0x400d0018 size=0x8d340 (578368) map[0m
[0;32mI (394) esp_image: segment 6: paddr=0x000cd360 vaddr=0x4008afa4 size=0x08658 ( 34392) load[0m
[0;32mI (408) esp_image: segment 7: paddr=0x000d59c0 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (409) esp_image: segment 8: paddr=0x000d5a2c vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (427) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (427) boot: Disabling RNG early entropy source...[0m
[0;32mI (430) cpu_start: Pro cpu up.[0m
[0;32mI (433) cpu_start: Starting app cpu, entry point is 0x40088e94[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (443) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (450) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (456) heap_init: At 3FFD6830 len 000097D0 (37 KiB): DRAM[0m
[0;32mI (462) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (468) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (475) heap_init: At 400935FC len 0000CA04 (50 KiB): IRAM[0m
[0;32mI (481) cpu_start: Pro cpu start user code[0m
[0;32mI (52) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (213) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0[0m
netif is not up, old style port?
NFPKTV1
```

josesimoes, 2018-10-07:
The deploy manager is looking for an assembly that is not available in your device image.
You are either referencing a mscorlib Nuget of v1.0.1.N or have the device flashed with an older image.
Could you please post here the device capabilities' output for checking?

ruslankiskinov, 2018-10-07:
The nanoFramework version is:
nanoFramework.CoreLibrary.1.0.0-preview073

The EspFirmwareFlasher is getting the image automatically and I don't know how to control which version to install and which version is compatible...

josesimoes, 2018-10-09:
@ruslankiskinov please give it another try. With the latest versions of everything (class libs & target image) you shouldn't be seeing this issue anymore.

josesimoes, 2018-10-11:
Lacking feedback from the developer, I'm closing this. 🙉 
Feel free to get back to this at any time.

closed 2018-10-11 by josesimoes

ruslankiskinov, 2018-10-11:
Sorry I was out of the country, but before that I tried couple of images of the firmware starting from 1024 to 1021 and 1021 was working fine. It appears that for  nanoFramework.CoreLibrary.1.0.0-preview073 the 0.1.0-preview.1021.image.zip version was the right one. 

I'll try with the latest version as you said and will let you know.

Thank you for your help!

josesimoes, 2018-12-04:
@jpearsonduluth thank you very much for writing this. And yes you are correct: right now this feature is undocumented. Adding it to the todo list... 😉