nanoframework/Home #421: Debugging I2C with single-stepping fails second time onwards (on STM32)

sharmavishnu, 2018-10-10:
### Details about Problem
Instantiate I2C device using the following code (address is for illustration only):
var i2cDevice = I2cDevice.FromId("I2C1", new I2cConnectionSettings(0x3C) { BusSpeed = I2cBusSpeed.StandardMode , SharingMode= I2cSharingMode.Shared });

Perform read/write with the peripheral and then call i2cDevice.Dispose();
Start debugging on visual studio and put a breakpoint on the line that instantiate the i2cDevice. Keep doing single-stepping (F10). For the first time, the single-stepping works. After program exits, start debugger again, but this time, after you press F10 when you reach the line that instantiate the I2C device, the debugger hangs for about 10 seconds and then exits. The code on the MCU continues to execute further. 
After the first successful debug, single-stepping, the next time onwards, it always hangs at the line where I2C device is getting instantiated. It blocks for about 10 seconds and debugger exits. This is consistent.

Only way to bring it back is to re-flash the device with the hex files and the process repeats (meaning, first time single-stepping goes through and next time onwards fails)

**nanoFramework area:** Visual Studio Extension/Debugger

**VS version:** Enterprise 2017, V 15.8.2

**VS extension version:** 0.5.0.49

**Target:** I have tried this on STM32F401RE

**Firmware image version:** Built with latest codebase as of yesterday (09-Oct-2018)

Worked before? If so, with which nanoFramework image version: Was working with codebase before from 18-Sep-18.

sharmavishnu, 2018-10-13:
Same issue observed while working with SPI. Debugger hangs for 10 seconds and then exits at the following line:
this._spiDevice = SpiDevice.FromId("SPI2", new SpiConnectionSettings(28) { ClockFrequency=8000000, Mode = SpiMode.Mode0  });

sharmavishnu, 2018-10-16:
@josesimoes Added new issue #424.

sharmavishnu, 2018-10-16:
New and Revised Observations:
**Correction** : The code in the MCU does not seem to continue to execute after the debugger hangs. When the debugger hangs, it stays there for about 10 seconds. Then it exists. It causes the MCU to reset and then the code executes correctly (since no debugger is attached). It seems, the 10 seconds is coming from the watchdog

**New Observations**: It seems there is a *hardfault* in single-stepping. When this happens, the only two ways to recover are:
1. Re-flash the booter and CLR hex
2. Erase the deployment and power cycle the MCU

If either of the above 2 are done, then single-stepping works again for once. 
**Observation on "Erase Deployment"**: If the deployment is erased using ST-Link (from menu Target > Erase Sectors, selected sector 6 and sector 7 for STM32F401RE), then the debugger works again, no need to re-flash the whole MCU with booter and CLR hrx files. However, when I do "Erase Deployment" from "Device Explorer", the debugger does not work. To enable debugging again for the particular I2C code, I must do "erase deployment" using ST-Link

StephenCardinale, 2018-11-07:
Confirmed to exhibit same bug using the Quail and STMF429i Discovery board. Except if have the same behavior without single-step debugging.

The I2C sample in the samples repro was tested on the STMF429i Discovery Board and using the Weather click on the Quail.

josesimoes, 2018-11-10:
Closed with nanoframework/lib-Windows.Devices.I2c#31

closed 2018-11-10 by josesimoes