nanoframework/Home #426: Extract System.Math to become an independent class lib

josesimoes, 2018-10-19:
## Motivation

This is to follow an old idea of being able to have _flavoured_ build which include only the required features/components. 
Extracting the "obvious" components/features empowers developers to use only what they need, thus saving flash space for app deployment. 

**This is a must for being able to use nanoFramework on more constrained devices!**

## Proof of concept

- Extracted the System.Math namespace from the ST_B_L072Z_LRWAN1 reference target.
- Build in MinSizeRel
- nanoCLR size:
  + before (as is today in the develop branch): 150392 Bytes
  + after: 131640 Bytes
  ❗️❗️ **TOTAL GAIN: 18752 Bytes or 12.5% space saving** 😯 

PS: there are other namespaces/classes in mscorlib that are serious candidates to be extracted as well.

josesimoes, 2018-10-20:
Internally (in mscorlib own code) three Math methods are being used: `Max`, `Min` and `Abs`.

❓❓ Would it be preferable to: ❓❓ 
1. Keep those there and just extract the others.
2. Provide the code internally and extract all the Math API.

steffalk, 2018-10-20:
If we need it in mscorlib itself, I vote for keeping it in the base package and only extract math functions that are really optional for the OS. Otherwise we had code duplicated, and that’s not a good thing.

 

cc @nanoframework/developers <https://github.com/orgs/nanoframework/teams/developers>

NemesisXB, 2018-10-22:
My vote is for 2. 
I agree with steffalk that duplication is bad, but I think splitting an already defined library is worse and `min`, `max`and `abs` are simple functions and should not take that much code. If they were more complex I would opt for 1.

AdrianSoundy, 2018-10-22:
I think you need to explain further what you intend to change.
The system.math implementation is in native code(c++) and the managed code for system.math is part of the mscorlib assembly. The space reduction is just from removing the managed code ?

From what i understand you are preposing is to remove the system.math manged code from mscorlib and have it as a separate assembly that would need to be incuded in every project that needs some maths. The native code will still exist in nanoclr and all the maths standard libraries will still be linked into nanoclr.

If you really want to make all space available when not using Maths functions then you should also look at removing the native code which would need to be a build option. 

The other way i see is you could have a mscorlib lite version which can be just used for resource constrained devices. Then you wouldn't need the separate System.Math

josesimoes, 2018-10-22:
The above size is for the nanoCLR firmware image only. The PE for mscorlib will, of course, be smaller than what it is now.

My proposal is to remove it from both the base class library _and_ native code (doensn't make sense to have without the other). Basically the linker takes care of removing what is not used. So, event if any call to C math exists anywhere in the code, that won't be harmed.

We won't ever get duplicate code.

Leaving `Max`, `Min` and `Abs` in the "standard" mscorlib is not that bad at all as those are probably the most used calls in System.Math.

I believe that we should look into this with an open mind and be willing to make some concessions on what could be seen as the usual and formal way of dealing with these matters.

I suggest that we have to reach an agreement on:

1. Does this makes sense?
2. Is it OK to split the System.Math namespace? (leaving `Max`, `Min` and `Abs` in mscorlib)

I'm OK with both because:
- We have to squeeze mscorlib if we intend to support smaller targets (which we should).
- Doing so should has to introduce the minimul possible changes to larger platforms and the way it currently works.
- Splitting features in separate NuGets is perfectly in order with what we've been doing until now and keeps both constrained and feature rich targets happy. Just need the extra step of add more NuGets (not such a big deal).
- Spliting namespaces is not a sin, even in the full .NET. There are several assemblyes that share namespaces and provide related features. Just live in different assemblies. 

Comming up with a reduced and self-contained mscorlib for smaller devices will probably be more dificult and harder to achieve an agreement on what should be there or not... Splitting it in separate packages addresses this dificuty too and adds flexibility to the platform as anyone is free to choose what is added to it's project.

MatthiasJentsch, 2018-10-22:
For me it's OK to remove the Math part and split the System.Math namespace.

After that we can build the nanoCLR with an additional option like "NF_FEATURE_MATH" : "ON" and have the Math part included. By default Math is not included in the native image. Then it's like any other feature.

Maybe in the future we can provide a service where a manufacturer of a nanoFramework application can check what will be included in his nanoCLR image. And then the service build's the image with the checked features included.

josesimoes, 2018-10-24:
I'm moving forward with this. From all the above I think we can accomplish:
1. Extracting the Math features and associated baggage. (the trigonometric and algebraic calls are responsible for the bulk of the C code there)
2. Leave those absolute minimum methods in mscorlib (this will take us closer to a lite version of it - as @AdrianSoundy mentioned - and has been discussed before)
3. This makes way to do a similar refactoring on other areas that are begging to be extracted, like reflection, converters and text APIs.

josesimoes, 2018-10-26:
Update on this: turns out that partial classes can't live on different assemblies... so there goes the approach of keeping abs, min and max on mscorlib and extract just the remaining ones.
I'm extracting all System.Math and leaving behind those three as internals.

josesimoes, 2018-10-26:
## Final numbers on this change 🔢 ⚖️ 

### DLLs (irrelevant for the purpose of reducing the deployment size, but still interesting)

#### mscorlib
- was 77.312 bytes, is now 76.800 bytes

#### System.Math
- is 5.632 bytes

### PEs (relevant for the purpose of reducing the deployment size)

#### mscorlib ⬇️
- was 38.784 bytes, is now 38.004 bytes, that's **less 524 bytes**  

#### System.Math ⬆️
- is now 1.100 bytes  

### nanoCLR (relevant for the purpose of reducing the deployment size)

#### STM32F769I_DISCOVERY (System.Math ON = equivalent functionality) ⬇️
- was 404.532 bytes, is now 396.328, that's **less 8.204 bytes** (😯 wasn't expecting a reduction, rather a tiny penalty, so this must be the result of some optimization the linker is able to do now)
- overall gain in deployment size: -524 +1.100 - 8204 = **7.628 bytes** 🤔 👍 ✔️  

#### STM32F769I_ST_NUCLEO64_F091RC (System.Math OFF) ⬇️
- was 164.716 bytes, is now 146.000 bytes, that's **less 18.716 bytes**
- overall gain in deployment size: - 524 - 1.8716 = **19.240 bytes** 👍 ✔️ 

#### ESP32 (couldn't find any output showing the image size... 🤔 )

closed 2018-10-26 by josesimoes