nanoframework/Home #430: Q: support .net standard

grahamehorner, 2018-11-10:
IMHO It would be great for the nanoframework to support the .net standard and build upon the great work of the mono and .net core teams. This would allow for the .net native ahead of time compiler to target/link with ESP32 libraries

grahamehorner, 2018-11-10:
I definitely would be interested in helping out, as IHMO .net standard and. net core has some BIG advantages over the micro framework not only as the core clr is open source, but that this would allow development on Linux, Mac and Windows using there favourite tools or VS.Code or Visual Stuido which in turn would/could attract more interest and a wider range of developers that could contribute with other microprocessor implementations

josesimoes, 2018-12-17:
@grahamehorner any update on this?

grahamehorner, 2018-12-17:
@josesimoes still wip had to delay due to hospitalised with broken hip

josesimoes, 2018-12-17:
😯  sorry to hear that! Wish you a speedy recovery.
Let's wait to better days to resume this. 😃 

On ELL, yes I've started watching the episode about that on Chanel 9 IoT show. Looks promising. We should tap into that!

grahamehorner, 2018-12-17:
Idea: is that is the ELL was. Support and had the ability to access WiFi, GPIO 
 etc it would a good basis for a headless platform #IoT https://www.github.com/dotnet/iot

grahamehorner, 2018-12-17:
@josesimoes thanks, hospitalisation is helping me get the time to research and dig deeper ✌️

grahamehorner, 2018-12-17:
FYI 
https://github.com/Microsoft/ELL/issues/78

grahamehorner, 2018-12-17:
project NELL aka Nano Embedded Learning Library for [ESP32, ... ] 😎

grahamehorner, 2018-12-17:
@josesimoes would you like to DM me would love to have a repo to play/discuss possible ELL or NELL project here.

Cheers
Grahame

josesimoes, 2019-02-21:
Guess this has been discussed and answered. Can be reopened later if needed.

closed 2019-02-21 by josesimoes

rayperkins, 2019-06-19:
Hi team, has there been any more traction on the .netstandard front?

josesimoes, 2019-06-21:
@rayperkins I'm afraid not... We are following the original plan and haven't made any other attempts towards this.

grahamehorner, 2019-06-21:
@rayperkins this is still on my radar and I'm almost back to health after my injury ready to start this in ernest as a side project. If your interested in contributing/helping make this happen please let me know.  likewise if anyone else interested please contact me @josesimoes also please let me know you thoughts, I value your input.

rayperkins, 2019-06-21:
@grahamehorner, great to hear that you're on the mend and plan to continue looking into this.
Firstly, I'll admit that I only have the a very vague idea for what is required. I suspect that it a matter of building in whatever is missing between nanoframework and the .netstandard API spec.

Furthermore, we'd have to assess if it is possible to get a workable implementation that is small enough for micro-controllers. We may find that nanoframework would grow into something that is only suitable for micros with large flash sizes (like the ESP32) and therefore make it unusable for smaller devices. Perhaps looking into the runtime sizes from the Microsoft Blazor project would be a good indicator (I think it is currently around 2MB). If it is not achievable, then perhaps a subset API 'wishlist' (like LINQ) can be the goal. 

Perhaps supporting just a useful subset is what the original plan is, is this correct @josesimoes?

I am keen to contribute and after reading through the issue list I'll put my hand up to help out with updating some docs and getting a testing strategy in place. I think this would be a good start as it would be difficult to build in a whole heap of new API's without some sort of testing pattern.

grahamehorner, 2019-06-21:
@rayperkins totally agree, I'm watching Blazor with interest as the 'tree shaking' and the AOT compiler improve, a thought I had was if a WASM runtime was small enough for a microcontroller and the 'tree sharking' AOT compiler (in development by both mono team and microsoft .net core teams) produced small packages then creating a microcontroller wasm runtime may be an option.

It's always a question of how small and performant you get the code; I believe optimizations in the .net core framework with Span<T> and ReadOnlyMemory<T> etc. brings to the table better memory management which again is important for microcontrollers.

I believe selection to the correct namespaces for the microcontroller is key here; and while this would not 100% .netstandard 2.0 it would be a subset build on the standard and possible lend itself to a laying of other .netstandard build components/libraries for specific use-cases.

josesimoes, 2019-06-21:
I'm happy to see interest (and will?) to move forward with this! 👏 😄 

The initial accessment on this matter was that AOT wasn't interely achievable (or made sense) for the MCUs we are targeting.
Reworking our execution engine to be able to _chew_ the raw assemblies produced by Roslyn is out the question because of the size of those. Not to mention that the extra processing would make the perfomance unusable.

We would like to keep our focus on the same range of MCUs that we are currently targeting. For more powerfull MCUs there are other offers available.

If this is a matter of API, that can be expanded without having to rewrite everything from the ground up!
I would say that the biggest blocker here is the lack of support for generics...

Anyway, suggest that we discuss this on our [Discord channel](https://discord.gg/gCyBu8T) as it's better suited for conversations, unlike GitHub issues. 😉

sake402, 2020-11-15:
Hi. @josesimoes.
You don't know how glad I am to finally find out that nanoFramework exists. I have been a C# and C/C++ for a long time, and I would really love to be able to develop for embedded in C#.

And the .netstandard conformance opens another  lots of possibilities for nanoFramework.

I will be very glad to contribute actively to this, only I don't have so much idea on how it all fits together currently.

josesimoes, 2020-11-15:
Hey @sake402 glad that you're happy for finding the project. 🙂
And happier that you're willing to help. We sure can use all the talent and enthusiasm that people are willing to pour into nanoFramework.
I see that you're already on Discord, so let's continue this conversation there.

sake402, 2020-11-15:
Hi again. I must confess, I have a very vague idea on how .netstandard work.
But to my understanding, it is some specific api that is expected to be implemented (natively or not) on every specific targets like .netcore, .netframework and now .nanaoframeworks.

Nothing stops a couple of this implemetation from being handled in c# rather than native, especially things like System.Linq. It is however the job of the runtime to bind this api call from code to their implemetation on the target when the code is being jitted on the target.

What I can seem to wrap my head around however is that when I compile a code with nanoFramework, does it upload IL bytecode or native ESP32 assembly code.

If it is byteCode, then I assume you already have an IL interpreter in the nf-interpreter binary.
If not, then we probably need to start developing a portable and fast IL interpreter that read the byteCode from flash and execute that in a loop.

Please correct me if I am mincing things up.

sake402, 2020-11-15:
So I dug through the source code and I see you already have Interpreter.cpp for interpreting the IL.

Can you let us know the barrier for .netstandard then?

josesimoes, 2020-11-16:
@sake402 this converstation fits on the developement-interpreter channel. But don't let this prevent you from engaging a conversation there.

Yes, we have an interpreter and an execution engine. 
Yes we are alredy "executing" IL.

A nanoFramework application goes through the same build steps as any other C# code.
After Roslyn has done it's thing we call our meta data process, which tweaks the generated IL and produces a shortened version of it, wchich is loaded into the nanoFramework devices.

We had to roll our own project system because: 1) the VS extensibility API didn't allow us to have language sub-projects and 2) we can't simply create a new framework target because those are hardcoded everywhere (from the project system, VS, NuGet, etc).

sake402, 2020-11-16:
Forgive my ignorance, but I am trying to understand half of what you said up here.
When a nanoFramework project is compiled by Roslyn, it generates either a full or an exe, whose il got tweaked. Now I could imagine adding a . netstandard reference to the same project and performing the same tweaking on it like every other assembly.

josesimoes, 2020-11-16:
@sake402 no need to apologise! You're trying to understand this, I get that.

You're going too fast there. Again:
1) Roslyn build the EXE or DLL as with any other C# project.
1.1) EXE or DLL outputs are produced as with any other C# project.
2) Our metadaprocessor is then feed with these outputs and a reduced PE file is produced, which is loaded into the MCU or execution.

> Now I could imagine adding a . netstandard reference to the same project and performing the same tweaking on it like every other assembly.
That won't work because we have our own mscorlib (which is a reduced version of the full .NET one) and that's the one we have to reference to compile against.

alberk8, 2023-07-18:
> > @rayperkins totally agree, I'm watching Blazor with interest as the 'tree shaking' and the AOT compiler improve, a thought I had was if a WASM runtime was small enough for a microcontroller and the 'tree sharking' AOT compiler (in development by both mono team and microsoft .net core teams) produced small packages then creating a microcontroller wasm runtime may be an option.
> 
> I think this is the obvious way forward. There are, as of now, multiple WASM runtimes that run on hardware such as the ESP32 (WASM3, WAMR, maybe others). All we need from here is a way to get concise WASM out of our toolchain.
> 
> There is the "bflat" toolchain, which (to my understanding) wires up Roslyn and NativeAOT in such a way that the output doesn't depend on the runtime (even the GC, depending on how one uses it). Maybe we could work like that, and use nanofw's GC in WASM (at least, until WASM gains its own GC)?

I don't think the approach of going to WASM is not feasible, you are just adding more code and memory on top of existing infrastructure.  If you want to go complete native then I would suggest Rust Lang.  Going across multiple MCU architecture is hard when you have to maintain the same API and language.   Btw .Net NativeAOT is not complete, it even does not support ASPNet MVC and not to mention majority of the nuget packages out there.  Bflat looks just like another experiment.

godefroi, 2023-07-18:
When you say this:

> NativeAOT is not complete, it even does not support ASPNet MVC and not to mention majority of the nuget packages out there.

What do you mean? Is NativeAOT unable to compile these packages for some reason? Why not?

I would assume that the biggest issue with NativeAOT for ESP32 would be that there's no NativeAOT compiler for either the Extensa ISA or the RISC-V ISA.

alberk8, 2023-07-19:
> When you say this:
>
>
> What do you mean? Is NativeAOT unable to compile these packages for some reason? Why not?
>
> I would assume that the biggest issue with NativeAOT for ESP32 would be that there's [...]

Even with the final release of NativeAOT, I am pretty sure that AspNet MVC will not be support as for the nugets out there you need to prepare for [trimming](https://learn.microsoft.com/en-us/dotnet/core/deploying/trimming/prepare-libraries-for-trimming) and not to use reflection.  AOT means that the compiler needs to know what you intent to do at compile time instead of figuring out at runtime (which CLR is very good at).   

To be able to compile C# to native code for each MCU architecture, would be very hard.  An analogy would be to fit a car with a petrol engine but the fuel used must be diesel.  Will it work, maybe but there will be a lots of workaround and tons of modifications.

godefroi, 2023-07-19:
I think it's important to be clear; as I understand it, it's not that NativeAOT doesn't support ASP.Net MVC, it's the other way around: ASP.Net MVC doesn't support NativeAOT, because it uses language and runtime features that cannot currently be used with NativeAOT.

Compiling C# to native code for each MCU architecture is no different fundamentally than compiling C# to native code for different operating systems and CPU architectures; you need a compiler for each architecture, and we don't have that. What we *do* have, however, is a .net runtime built for WASM, and there are many WASM runtimes that run on all the interesting microcontroller ISAs (everything from AVR to various ARM flavors, Extensa, MIPS, RISC-V, others), so it seems to me like much of the work is done for us!

Ellerbach, 2023-07-20:
> Compiling C# to native code for each MCU architecture is no different fundamentally than compiling C# to native code for different operating systems and CPU architectures

There is a difference: the interop part on which nano is build on. All the interpreter is native and specific to a platform. Yes, that *could* be abstracted into another engine.
Then all the native elements like GPIO, I2C, etc are also specific, the interop mechanism is the same as for the rest. Each WASM would have to give this native connection in a standard way, which is not the case. So if it's to ru an hello world because you're blocked into your runtime, then it's a bit useless.
That is what makes things quite complicated, not the code itself.

godefroi, 2023-07-20:
> All the interpreter is native and specific to a platform.

Right, but we're talking about native compilation; there's no interpreter.

Ellerbach, 2023-07-21:
> Right, but we're talking about native compilation; there's no interpreter.

And that's one of the key point: you will have to create all the plumbing that goes with it to access the GPIO, I2C, etc. They are all different in all the MCU even in the same family.