nanoframework/Home #433: Visual Studio hang when creating new nanoFramework Project + workaround

fivefishstudios, 2018-11-19:
Hi all, just new to nanoframework and trying to follow tutorial and discovered an issue, plus a workaround. Wonder if this issue is just local to me or a known one?

Using Visual Studio 2017 15.9.1 
Installed nanoFramework VS2017 Extension v1.0.1.0  

Trying to follow the Getting Started tutorial... 

STEPS TO REPRODUCE PROBLEM
-----------------------------------
Open VStudio 2017.
Create a new Project using the nanoFramework as the template
VStudio 2017 just hangs and need to be shutdown using Task Manager

WORKAROUND
---------------
Open VStudio 
Create a new Project, using .NET Core Console App
Close Solution/Project
Now create a new Project using nanoFramework
... task done in seconds, and new nanoFramework project is created successfully

You can close this Project and create further new nanoFramework projects and there'll be no hang-ups. 

It seems the first time you open VStudio, you cannot create a new nanoFramework Project without creating a dummy .NET Console app first.

josesimoes, 2018-11-19:
Hi @fivefishstudios and welcome to nanoFramework! 😃 

Yes, other devs have already reported this behaviour after installing VS15.9.
We'll be releasing an update for the extension to take care of this. 

In the mean time, thank you very much for providing a workaround for folks that are already on that VS version. 👏

josesimoes, 2018-12-17:
Status update: this has been reported to VS developer community [here](https://developercommunity.visualstudio.com/content/problem/405079/can-start-a-new-nanoframework-project-from-a-templ.html).

Still waiting for feedback from the VS team. 
Suggest that devs using the extension subscribe to that issue and upvote it.

josesimoes, 2019-01-10:
Updated my VS to 15.9.5 and the issue remains: starting a project from a nF template hangs VS.
Apart from that, everything seems to be working as expected.

josesimoes, 2019-01-22:
🎉 🎉 🎉  Great news: the status of this issue in VS website was changed to **"Fixed - Pending Release"**

along with the comment:

"We have fixed the problem in an upcoming release. Thank you for your feedback!"

josesimoes, 2019-01-25:
Fixed in VS [15.9.6](https://docs.microsoft.com/en-us/visualstudio/releasenotes/vs2017-relnotes#15.9.6).

closed 2019-01-25 by josesimoes