nanoframework/Home #439: ESP32 wifi fail on preview 152

BucurCosmin, 2018-12-03:
Looks like on the preview 1.0.2 build 152 the wifi on esp32 devkitc fails.
On released 1.0.1 works ok.
I was trying the preview version since I have an issue on sockets. Listening fails cause of backlog.

Cosmin

josesimoes, 2018-12-03:
Hi @BucurCosmin 

Please give it another try with the latest preview version (152 is not the latest one).

Anyways someone from @nanoframework/esp32-developers will check this.

BucurCosmin, 2018-12-03:
Hi @josesimoes 
I tried with the last build 182 but deployment is failing.

2>nanoDevice is initialized
2>Deploy failed.
2>***************************************
2>Couldn't find a valid native assembly required by System.Net v1.0.2.13, checksum 0xC58744CD.
2>Couldn't find a valid native assembly required by mscorlib v1.0.6.10, checksum 0x49ABE593.
2>Couldn't find a valid native assembly required by nanoFramework.Runtime.Events v1.0.2.17, checksum 0xBFF88292.
2>Couldn't find a valid native assembly required by nanoFramework.Hardware.Esp32 v1.0.2.12, checksum 0xF8079179.
2>Couldn't find a valid native assembly required by Windows.Devices.Wifi v1.0.2.11, checksum 0x1B4DDB17.
2>Couldn't find a valid native assembly required by Windows.Devices.Gpio v1.0.2.15, checksum 0xF8E84C07.
2>***************************************
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========

BucurCosmin, 2018-12-03:
Actually the packages are referenced correctly. Maybe there is an issue with the extension. I'm using visual studio 15.9.2

josesimoes, 2018-12-03:
Yes, there are known issue with VS 15.9 see #433

BucurCosmin, 2018-12-05:
@josesimoes even with build 186 wifi still not working on esp32

josesimoes, 2018-12-05:
@BucurCosmin lets check a couple of things to get this straight and be able to help you get past this. 
Please post here the following (these were requested on the issue template):

- [ ] Device capabilities reported by VS
- [ ] The packages.config file content
- [ ] The output of the deploy stage on VS output window
- [ ] A screenshoot of the solution explorer on your VS

BucurCosmin, 2018-12-06:
@josesimoes 
**device capabilities:**
System Information
HAL build info: 1.0.2.186, nanoFramework running ESP32 built with IDF v3.1

Image build @ Dec  3 2018 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.0.2.186, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:
  NFApp3, 1.0.0.0
  System.Net, 1.0.2.23
  Windows.Devices.Wifi, 1.0.2.17
  Windows.Devices.Gpio, 1.0.2.25
  mscorlib, 1.0.6.21
  nanoFramework.Runtime.Events, 1.0.2.23

Native Assemblies:
  mscorlib v1.0.6.21, checksum 0x7CCFBEC1
  nanoFramework.Runtime.Native v1.0.2.16, checksum 0xFE14C7A7
  nanoFramework.Hardware.Esp32 v1.0.2.17, checksum 0xF8079179
  nanoFramework.Networking.Sntp v1.0.2.49, checksum 0x733B4551
  nanoFramework.Runtime.Events v1.0.2.23, checksum 0xBFF88292
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v1.0.2.15, checksum 0x4BDCF00F
  System.Net v1.0.2.23, checksum 0xC58744CD
  Windows.Devices.Adc v1.0.2.24, checksum 0xD21F3E86
  Windows.Devices.Gpio v1.0.2.25, checksum 0xF8E84C07
  Windows.Devices.I2c v1.0.2.25, checksum 0x0C4EA1B2
  Windows.Devices.Pwm v1.0.2.17, checksum 0x84B1FBF8
  Windows.Devices.SerialCommunication v1.0.2.13, checksum 0x093798D6
  Windows.Devices.Spi v1.0.2.19, checksum 0x3DE083A4
  Windows.Devices.Wifi v1.0.2.17, checksum 0x1B4DDB17

--------------------------------
::        Memory Map          ::
--------------------------------
Type     Start       Size
--------------------------------
RAM    0x200000c0  0x00020000
FLASH  0x08000000  0x00080000

-----------------------------------------------------------
::                   Flash Sector Map                    ::
-----------------------------------------------------------
Invalid or empty map data.

Deployment Map
Empty

**Packages config:**
<?xml version="1.0" encoding="utf-8"?>
<packages>
  <package id="nanoFramework.CoreLibrary" version="1.0.6-preview-021" targetFramework="netnanoframework10" />
  <package id="nanoFramework.Runtime.Events" version="1.0.2-preview-023" targetFramework="netnanoframework10" />
  <package id="nanoFramework.System.Net" version="1.0.2-preview-023" targetFramework="netnanoframework10" />
  <package id="nanoFramework.Windows.Devices.Gpio" version="1.0.2-preview-025" targetFramework="netnanoframework10" />
  <package id="nanoFramework.Windows.Devices.Wifi" version="1.0.2-preview-017" targetFramework="netnanoframework10" />
</packages>

**deployment:**
2>------ Deploy started: Project: NFApp3, Configuration: Debug Any CPU ------
2>Getting things ready to deploy assemblies to nanoFramework device: Silicon Labs CP210x USB to UART Bridge (COM8) @ COM8.
2>nanoDevice is initialized
2>Adding NFApp3 v1.0.0.0 (2844 bytes) to deployment bundle
2>Adding System.Net v1.0.2.23 (14524 bytes) to deployment bundle
2>Adding Windows.Devices.Wifi v1.0.2.17 (3680 bytes) to deployment bundle
2>Adding Windows.Devices.Gpio v1.0.2.25 (4420 bytes) to deployment bundle
2>Adding mscorlib v1.0.6.21 (38272 bytes) to deployment bundle
2>Adding nanoFramework.Runtime.Events v1.0.2.23 (2624 bytes) to deployment bundle
2>Deploying 6 assemblies to device... Total size in bytes is 66364.
2>Deployment successful.
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========

josesimoes, 2018-12-06:
It seems that the deployment is successful... guess you're past the deployment issues now.
So I guess this closes the issue.

The real issue now is that socket one that you've mentioned before. Is this correct?

josesimoes, 2018-12-06:
Could you be more specific about what exactly "not working" means?

BucurCosmin, 2018-12-06:
It's not connecting to the wifi router

josesimoes, 2018-12-06:
OK. That's a better description! 👍 
More details that you can share? Maybe posting some code?

❓ Have you tried with the [Wi-Fi samples](https://github.com/nanoframework/Samples/tree/master/samples/Wifi) ❓ 

cc @nanoframework/esp32-developers

BucurCosmin, 2018-12-06:
The code is (I removes the actual SSID and pass:
`using System;
using System.Threading;
using Windows.Devices.WiFi;
using Windows.Devices.Gpio;
using NFApp3;
using SimpleWebServer;
namespace ScanWiFi
{
    public class Program
    {
        // Set the SSID & Password to your local WiFi network
        const string MYSSID = "blabla";
        const string MYPASSWORD = "blabla";
        private static bool connected = false;
        public static void Main()
        {
            
            try
            {
                // Get the first WiFI Adapter
                WiFiAdapter wifi = WiFiAdapter.FindAllAdapters()[0];

                // Set up the AvailableNetworksChanged event to pick up when scan has completed
                wifi.AvailableNetworksChanged += Wifi_AvailableNetworksChanged;

                // Loop forever scanning every 30 seconds
                while (connected==false)
                {
                    
                    wifi.ScanAsync();

                    Thread.Sleep(10000);
                }
            }
            catch (Exception ex)
            {
               
            }
            new Thread(SimpleWebServer.WebServer.StartListening).Start();
            while (true)
            {
                Thread.Sleep(5000);
            }

            Thread.Sleep(Timeout.Infinite);
        }

        /// <summary>
        /// Event handler for when WiFi scan completes
        /// </summary>
        /// <param name="sender"></param>
        /// <param name="e"></param>
        private static void Wifi_AvailableNetworksChanged(WiFiAdapter sender, object e)
        {
            Console.WriteLine("Wifi_AvailableNetworksChanged - get report");

            // Get Report of all scanned WiFi networks
            WiFiNetworkReport report = sender.NetworkReport;

            // Enumerate though networks looking for our network
            foreach (WiFiAvailableNetwork net in report.AvailableNetworks)
            {
                // Show all networks found
                //Console.WriteLine($"Net SSID :{net.Ssid},  BSSID : {net.Bsid},  rssi : {net.NetworkRssiInDecibelMilliwatts.ToString()},  signal : {net.SignalBars.ToString()}");

                // If its our Network then try to connect
                if (net.Ssid == MYSSID && connected==false)
                {
                    // Disconnect in case we are already connected
                    sender.Disconnect();

                    // Connect to network
                    WiFiConnectionResult result = sender.Connect(net, WiFiReconnectionKind.Automatic, MYPASSWORD);

                    // Display status
                    if (result.ConnectionStatus == WiFiConnectionStatus.Success)
                    {
                        connected = true;
                    }
                    else
                    {
                        //Console.WriteLine($"Error {result.ConnectionStatus.ToString()} connecting o Wifi network");
                    }
                }
            }
        }
    }
}
`

josesimoes, 2018-12-06:
Linked to this in Discord#esp32. You may want to follow up there.

closed 2018-12-08 by josesimoes

ishvedov, 2018-12-09:
With build 195 everything goes ok