nanoframework/Home #440: Update ESP32 firmware flasher documentation

josesimoes, 2018-12-04:
Update the readme (and possibly the getting started docs too) with the new features since it was initially launched.

MatthiasJentsch, 2018-12-04:
There is nothing to update. The latest features are documented since 3 month here: https://github.com/nanoframework/nf-tools/tree/master/EspFirmwareFlasher also the option **-t** that was mentioned in #419 is documented. I think we can close this.

jpearsonduluth, 2018-12-06:
@MatthiasJentsch and @josesimoes , Thank you for looking into this. Indeed the EspFirmwareFlasher is documented, and quite nicely. However, the [How to flash nanoFramework on an ESP32](https://www.youtube.com/watch?v=WrPg8CLTr0Y) youtube video points to "Releases" page of this repository, which makes no reference to the documentation. Adding a note to the releases page about where to find the documentation would be very helpful IMHO. Thanks for all your hard work on this!

josesimoes, 2018-12-06:
You're most welcome and thank you for you words! 😃 

I've edited the video description to include a link to the tool documentation. Hopefully that will help others bumping into this.

MatthiasJentsch, 2018-12-06:
@josesimoes Thanks for editing the video description. Btw it's also self explaining when you call the tool with the **--help** option.