nanoframework/Home #441: IoT GPIO, I2C, PWM, SPI ...

grahamehorner, 2018-12-05:
IMHO the nanoframework should follow the API created by the core CLR/FX for IoT see: https://github.com/dotnet/iot this would allow developers to prototype/debug using
hardware such as the RaspberryPI and/or share code between different OS and hardware
platforms.

josesimoes, 2018-12-05:
Hi @grahamehorner!

I've compared the GPIO and SPI classes and I believe we overlap each other on most of it and are pretty close on the other... This must be because we are following the UWP API and I believe they are doing pretty much the same. 

Anything in particular that we are deviating too much from the original?

PS: wasn't aware of this project, thanks for the heads up 👍

grahamehorner, 2018-12-05:
@josesimoes It would IMHO be great to align the namespaces and structure with possibly a wrapper to transform to newer style api in .net core into that which was exposed in the microframework, thus allowing development to move forward in terms of newer .net core standards while also providing support for the older framework.

josesimoes, 2018-12-05:
We are in a good shape to do that (because of the overlap and similarities).
As they state on their repo, the API is most likely changing. We are better waiting for a stable version and then move into making the required changes. 

We the current limited resources it makes no sense keeping up with their changes. Unless, of course, someone is willing to take care of this. It's not that hard, actually... 😉

grahamehorner, 2018-12-05:
@josesimoes I'm happy to do this; IMHO I think the view that this would bring out weighs any pain that maybe experienced with the changing API surface.

josesimoes, 2018-12-06:
Understood. Let's open the discussion about this in [Discord#General](https://discord.gg/gCyBu8T) and ask the community about this.

josesimoes, 2018-12-17:
@grahamehorner not much feedback from the community about this. Nevertheless this looks like the path to follow. 

Are you still interested in helping out? If yes we can start a POC for GPIO has it shouldn't be that complicated and we'll see here that takes us.

josesimoes, 2018-12-17:
I've filled an issue on the dotnet/iot repo proposing an improvement (which nanoframework already has 😉 🚀 ) provide a Toggle() method for Gpio pins. 

Check [here](https://github.com/dotnet/iot/issues/107).

grahamehorner, 2018-12-17:
@josesimoes definitely, also looking at some kind of system where you can create Virtual/simulated electronic circuit that are connected to the GPIO, PWM, drivers.

grahamehorner, 2018-12-17:
@josesimoes would you like to DM me would love to have a repo to play/discuss possible ELL or NELL project here.

Cheers
Grahame

josesimoes, 2018-12-17:
@grahamehorner just pinged you on Discord

grahamehorner, 2018-12-17:
Ok, unfortunately the hospital WiFi firewall is blocking 😩

CAUTION: This email and any attachments transmitted with it contain information that is CONFIDENTIAL and may be privileged or otherwise protected from disclosure. If you are not the intended recipient please notify us immediately and delete all copies of this email message and any attachments transmitted with it from your system. If you are not the intended recipient, any use, disclosure or copying of this email message and any attachments transmitted with it is strictly prohibited. This email message and any attachments transmitted with it cannot be guaranteed to be secure or error free as information could be intercepted, corrupted, lost, destroyed or incomplete. Although we routinely screen for viruses, it makes no representation or warranty as to the absence of viruses in this email or any attachments. The views expressed in this email are those of the sender and not necessarily those of Gabriel Cloud Systems LTD; if you require clarification please contact the sender.

josesimoes, 2019-04-20:
Any update on this matter? If not, I'll close the issue and it can be reopened at any time.

josesimoes, 2019-05-06:
Seems that there are no updates. Closing issue.

closed 2019-05-06 by josesimoes