nanoframework/Home #447: SPI module malfunctioning with ST type boards

Dweaver309, 2019-01-28:
Please read the following information before posting the issue.

## Before posting the issue...

If you need clarification on something, need help on a particular situation or want to start a discussion, instead of opening an issue it is best to start a conversation on one of our Discord channels. 
Go there and select the channel that's most appropriate to the context so subject matter experts are most likely to answer promptly.
You can join our lively [Discord community](https://discordapp.com/invite/gCyBu8T)

:exclamation: Remove the content above here and fill out details below. :exclamation:

### Details about Problem

**nanoFramework area:** (C# code | Visual Studio extension | Hardware/target board | Nuget packages | Community targets | Community contributions | Documentation )

**VS version:** 

**VS extension version:** 

**Target:** 

**Firmware image version:** 

**Device capabilities output:** 

Worked before? If so, with which nanoFramework image version: 

### Detailed repro steps so we can see the same problem

1.

2.

...

### Other suggested things

### Sample Project (if applicable)

### Screenshot (if applicable/relevant)

Dweaver309, 2019-01-28:
I am trying the create a spi device on a Netduino 3 wifi board. 

The first time I run the sample program in VS the spi device is created. 

The next time I run the program it stops on the line:
 
_TestSPIDevice = SpiDevice.FromId("SPI2", connectionSettings) 

No error is given the application justs stops.

If I erase the application area and toggle power the spi device is created again.

Is there an issue with the spi module or is there something I need to put in my code?

[Program.zip](https://github.com/nanoframework/Home/files/2802558/Program.zip)

josesimoes, 2019-01-29:
@Dweaver309 it would be great if you could follow the issue template in future issues... it's there to help developers reporting issues and for the team to "parse" it more efficiently 😉

josesimoes, 2019-05-06:
Is this still an issue?

Dweaver309, 2019-05-06:
I think is issue has been corrected. I moved my project to an ESP32 board that works as intended.  I have seen code on Discord that is working. I am closing this issue.

closed 2019-05-06 by Dweaver309