nanoframework/Home #450: Documentation on limitations

Korporal, 2019-01-30:
Is there any documentation on what restrictions or limitations nanoframework has? For example can we write unsafe code or are generics supported and so on?

josesimoes, 2019-01-30:
I'm afraid there is no such document...
On what concerns the BCL we have a very good coverage when comparing with the standard .NET framework. Generics and async/await would be the outstanding features that we don't have support for.

On the class libraries we are following as much as possible the UWP API.

Considering that a nF project references its own mscorlib (and associated class libraries) it's supposed to be pretty much self describing from the Intellisense.

As for unsafe code yes, you can use that (at your own risk of course). It requires manualy editing the .nfproj file to add the respective property, as described [here](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/compiler-options/unsafe-compiler-option).

closed 2019-01-30 by josesimoes

acrigney, 2021-09-24:
Can we have an update on the generic support now guys? Async can wait I know that the way that .net full framework does this is probably not ideal for an embedded controller.