nanoframework/Home #452: VS loses contact with STM32 board after a physical board reset.

Korporal, 2019-02-03:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension)

**VS version: 2017 v15.9.6** 

**VS extension version: 1.0.3.32** 

**Target: STM32F7 DISCO Board with LCD** 

**Firmware image version: ST_STM32F769I_DISCOVERY-1.0.3** 

**Device capabilities output:** 

> 
> 
> 
> System Information
> HAL build info: 1.0.3.230, nanoCLR running @ ST_STM32F769I_DISCOVERY built with ChibiOS v18.2.1.2
> 
> Image build @ Dec 17 2018 GNU ARM GCC v7.3.1
> 
> OEM Product codes (vendor, model, SKU): 0, 0, 0
> 
> Serial Numbers (module, system):
>   00000000000000000000000000000000
>   0000000000000000
> 
> Solution Build Info: 1.0.3.230, nanoCLR running @ ST_STM32F769I_DISCOVERY built with ChibiOS v18.2.1.2
> 
> AppDomains:
> 
> Assemblies:
>   NFApp1, 1.0.0.0
>   nanoFramework.Hardware.Stm32, 1.0.2.2
>   mscorlib, 1.1.0.4
>   nanoFramework.Runtime.Events, 1.0.2.2
>   Windows.Devices.Gpio, 1.0.2.2
> 
> Native Assemblies:
>   mscorlib v1.1.0.4, checksum 0x7CCFBEC1
>   nanoFramework.Runtime.Native v1.0.2.2, checksum 0xFE14C7A7
>   nanoFramework.Hardware.Stm32 v1.0.2.2, checksum 0xFFC9E917
>   nanoFramework.Devices.Can v1.0.1.2, checksum 0xD2E9416A
>   nanoFramework.Devices.OneWire v1.0.1.2, checksum 0x4BFAF309
>   nanoFramework.Networking.Sntp v1.0.2.2, checksum 0x733B4551
>   nanoFramework.Runtime.Events v1.0.2.2, checksum 0xBFF88292
>   EventSink v1.0.0.0, checksum 0xF32F4C3E
>   System.Math v1.0.2.2, checksum 0x4BDCF00F
>   System.Net v1.0.2.2, checksum 0xFD8EBD9C
>   Windows.Devices.Adc v1.1.0.2, checksum 0xD21F3E86
>   Windows.Devices.Gpio v1.0.2.2, checksum 0xF8E84C07
>   Windows.Devices.I2c v1.1.0.2, checksum 0x0C4EA1B2
>   Windows.Devices.Pwm v1.1.0.2, checksum 0x84B1FBF8
>   Windows.Devices.SerialCommunication v1.0.2.17, checksum 0x093798D6
>   Windows.Devices.Spi v1.1.0.2, checksum 0x3DE083A4
> 
> 
> 
> --------------------------------
> ::        Memory Map          ::
> --------------------------------
> Type     Start       Size
> --------------------------------
> RAM    0xc0000000  0x01000000
> FLASH  0x08000000  0x00200000
> 
> 
> 
> -----------------------------------------------------------
> ::                   Flash Sector Map                    ::
> -----------------------------------------------------------
>   Region     Start      Blocks   Bytes/Block    Usage
> -----------------------------------------------------------
>       0    0x08000000       1      0x008000     nanoBooter
>       1    0x08008000       1      0x008000     Configuration
>       2    0x08010000       2      0x008000     nanoCLR
>       3    0x08020000       1      0x020000     nanoCLR
>       4    0x08040000       1      0x040000     nanoCLR
>       5    0x08080000       6      0x040000     Deployment
> 
> 
>  Start        Size (kB)           Usage
> --------------------------------------------
>  0x08000000   0x008000 (32kB)     nanoBooter
>  0x08008000   0x008000 (32kB)     Configuration
>  0x08010000   0x070000 (448kB)    nanoCLR
>  0x08080000   0x180000 (1536kB)   Deployment
> 
> 
> 
> Deployment Map
> Empty
> 

### Detailed repro steps so we can see the same problem

1. Build and deploy a simple app.

2. Ensure app is running (e.g. LED activity etc)

3. Reset the board (Black push button)

4. Visual Studio (Device Explorer) loses contact with board and never see it again unless VS is restarted.

### Screenshot (if applicable/relevant)

**Before restarting VS**

[image]

**After restarting VS**

[image]

josesimoes, 2019-02-05:
Can't reproduce this...

Some follow up questions:
1. Do you have another instance of VS running? Sometime this can prevent a board from showing on the other instances.
2. Can you hear the Windows sound of hardware connect/disconnect?
3. Can you please enable the debug messages (exclamation icon rightmost icon in Device explorer window) reproduce the event at your end and paste here the output of the nanoFramwork extension?

Korporal, 2019-02-19:
@josesimoes - Apologies for the delay. I shut all instances of VS then reopened the VS with a simple test NF project (that does deploy and rune fine unless VS gets this problem).

After restarting just a single instance of VS with the STM32 connected, here's what I see:

[image]

The I press the Start Debug green arrow and here's what happened next:

[image]

josesimoes, 2019-02-19:
From the log above, something is causing the device to reset after the deployment.

If you don't have any instruction requesting that, you may have some code causing an hard fault.
Try connect SWO output in STLink and see if anything interesting show up.

josesimoes, 2019-02-21:
If this happens you should now be able to use the new rescan button in Device Explorer.

closed 2019-02-21 by josesimoes