nanoframework/Home #457: No Device Explorer tool window

adigostin, 2019-02-10:
Trying to create my first nanoFramework app. Installed "nanoFramework VS2017 Extension". And then:
 * File -> New Project -> ... -> Blank Application (nanoFramework) -> Ok.
 * Build -> Build Solution -> … "1 succeeded."
 * Debug -> Start Debugging -> … -> "There is no device selected. Please select a device in Device Explorer tool window."
 * View -> Other Windows -> click on "Device Explorer". **Expected result**: some tool window shows up. **Actual result**: nothing happens; nanoFramework unusable.

josesimoes, 2019-02-10:
Hi @adigostin, welcome to nanoFramework!

Some important information is missing that was on the issue template:
- what is the version of the extension you have installed
- what is the VS version that you are using

Nevertheless let's try to help you. The expected outcome of showing the Device Explorer window is that a tool window (Device Explorer) shows up docked in the group of Solution Explorer. It's not a pop up of some window that is floating, 

Second, the "complain" about no device selected should be unrelated to the above.
After you manage to show Device Explorer, check if there is actually a device there.

adigostin, 2019-02-10:
Extension version: 1.0.3.54
VS version: 15.9.6
The "complaint" is not about the "no device selected" message; it is about the Device Explorer tool window not showing up, as can be seen in the issue title and the issue text.

josesimoes, 2019-02-10:
Two suggestions:

- please uninstall and reinstall the extension
- if the behaviour remains check VS activity log to find out any errors or warnings

josesimoes, 2019-02-11:
Please give it a try to v1.0.3.56 and report your feedback here.

adigostin, 2019-02-11:
The Device Explorer tool window works now with v1.0.3.56.

closed 2019-02-11 by adigostin

LeFouSAgite, 2019-05-21:
Hello,

I have exactly the same problem as above for adigostin "No device Explorer tool window".
VS 2017  15.8.1
nanoFramework 1.0.3.137

there are 2 errors in activitylog but i don"t know how to manage this ! Can someone help me please ?

<entry>
    <record>623</record>
    <time>2019/05/21 15:09:08.350</time>
    <type>Error</type>
    <source>VisualStudio</source>
    <description>SetSite failed for package [Visual Studio 2017 extension for nanoFramework. Enables creating C# Solutions to be deployed to a target board and provides debugging tools.]Source: &apos;nanoFramework.Tools.Debugger&apos; Description: Type Windows Runtime &apos;Windows.Devices.Enumeration.DeviceWatcher&apos; introuvable.&#x000D;&#x000A;System.TypeLoadException: Type Windows Runtime &apos;Windows.Devices.Enumeration.DeviceWatcher&apos; introuvable. ---&gt; System.PlatformNotSupportedException: L&apos;op&#x00E9;ration n&apos;est pas prise en charge sur cette plateforme.&#x000D;&#x000A;   --- Fin de la trace de la pile d&apos;exception interne ---&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.Debugger.PortSerial.SerialPort..ctor(Object callerApp, Boolean startDeviceWatchers)&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.NanoDeviceCommService.get_DebugClient()&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.ToolWindow.ViewModel.DeviceExplorerViewModel.OnNanoDeviceCommServiceChanged()&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.ToolWindow.ViewModel.DeviceExplorerViewModel.set_NanoDeviceCommService(INanoDeviceCommService value)&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.DeviceExplorerCommand.&lt;InitializeAsync&gt;d__23.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.&lt;&lt;InitializeAsync&gt;b__36_0&gt;d.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; Microsoft.VisualStudio.Threading.JoinableTask.&lt;JoinAsync&gt;d__78.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; nanoFramework.Tools.VisualStudio.Extension.NanoFrameworkPackage.&lt;InitializeAsync&gt;d__36.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; Microsoft.VisualStudio.Shell.AsyncPackage.&lt;&gt;c__DisplayClass17_0.&lt;&lt;Microsoft-VisualStudio-Shell-Interop-IAsyncLoadablePackageInitialize-Initialize&gt;b__1&gt;d.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; Microsoft.VisualStudio.Threading.ThreadingTools.&lt;WithCancellationSlow&gt;d__12.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; Microsoft.VisualStudio.Threading.JoinableTask.&lt;JoinAsync&gt;d__78.MoveNext()&#x000D;&#x000A;--- Fin de la trace de la pile &#x00E0; partir de l&apos;emplacement pr&#x00E9;c&#x00E9;dent au niveau duquel l&apos;exception a &#x00E9;t&#x00E9; lev&#x00E9;e ---&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)&#x000D;&#x000A;   &#x00E0; System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)&#x000D;&#x000A;   &#x00E0; Microsoft.VisualStudio.Threading.JoinableTask.CompleteOnCurrentThread()&#x000D;&#x000A;   &#x00E0; Microsoft.Visua [... cut ...]
    <hr>80131522</hr>
    <errorinfo></errorinfo>
  </entry>
  <entry>
    <record>624</record>
    <time>2019/05/21 15:09:08.366</time>
    <type>Error</type>
    <source>VisualStudio</source>
    <description>End package load [Visual Studio 2017 extension for nanoFramework. Enables creating C# Solutions to be deployed to a target board and provides debugging tools.]</description>
    <guid>{046B40EB-1DE1-4D08-AF61-FDB7592B9BBD}</guid>
    <hr>80131522</hr>
    <errorinfo></errorinfo>
  </entry>

josesimoes, 2019-05-22:
What is the Windows version you're running?

LeFouSAgite, 2019-05-22:
Win 7 .
I asked also on Discord because was not sure you will see my comment on this closed issue.
Thanks

LeFouSAgite, 2019-05-22:
win 7 pro version SP1
I installed NuGet packages from official NuGet Gallery

Also, I never succeded to get a com port after programming boot and clr on STM32 F429 DISCO

My colleague has the same DELL E6440 laptop as mine but updated to win10 pro and VS2019. I just make a quick trial by installing nF extension => same result , no device explorer window is opening and an error message asking to close and restart VS

josesimoes, 2019-05-22:
Just for the record and anyone else looking at this:
That is the reason why: the debugger component, specifically the USB/COM port layer is using UWP API which is only available on Windows 8 and above.

xNiux, 2019-11-13:
Hi, I have the same issue with the following version :
- VS2017 Express : 15.9.17
- NF Extension : 1.2.5.38
- Windows : 10 (17.09

[image]
)

josesimoes, 2019-11-13:
@xNiux see https://github.com/nanoframework/Home/issues/480#issuecomment-496212985

xNiux, 2020-01-03:
> Hi, I have the same issue with the following version :
> 
> * VS2017 Express : 15.9.17
> * NF Extension : 1.2.5.38
> * Windows : 10 (17.09)
> 
> [image]
> 
> I'm also trying the Blinky sample, and flash it with de nanoff, but the LED doesn't blinking ...
> [image]

Answer from Microsoft team : It is a bug from the VS2017 Express, normally it is not possible to install extension ....

josesimoes, 2020-01-03:
@xNiux join our Discord community [here](https://discord.gg/gCyBu8T)