nanoframework/Home #458: Building STM32F0 targets with ChibiOS 19.x requires GCC 5

josesimoes, 2019-02-13:
See the original post with this discussion on [ChibiOS forum](http://www.chibios.com/forum/viewtopic.php?f=35&t=4894&start=20).

Reported bug in GNU GCC ticketing [here](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1803508).

After an official release of GCC toolchain and the following fix in ChibiOS this won't be required anymore.
Until then building STM32F091 target with ChibiOS requires specifying the GCC toolchain version.

Changes in the Azure Pipelines yaml's implemented with nanoframework/nf-interpreter#1209.

The GCC version can be removed after the above is fixed.

josesimoes, 2019-08-12:
Still active

networkfusion, 2019-08-26:
GCC 9.2 has now been released which fixes this bug (see https://gcc.gnu.org/gcc-9/changes.html). We just need to wait for it to make its way into the ARM toolchain and then ChibiOS.

networkfusion, 2020-05-12:
GCC 10.1 has been released! just waiting on ARM (hopefully)! For future reference, note the following: ```GCC 10 changed the default to the stricter -fno-common which causes errors when multiple static variables with the same name is defined in multiple compilation units.```

networkfusion, 2020-06-06:
No longer blocked if we move to the latest GCC version:

[image]

josesimoes, 2020-06-06:
If the extra check with ChibiOS is still there, we need to add it to our build definitions

josesimoes, 2020-06-08:
Confirmed! We are good to go with this one. 🚀

closed 2020-06-29 by josesimoes