nanoframework/Home #459: Inexplicable deploy problem

Korporal, 2019-02-19:
If low level tech detail are required I can provide them but basically I updated a simple STM32 DISCO project that ran OK, by adding a Nuget package for Serial IO:

[image]

Left is before, right is after.

The when I try to deploy I see:
[image]

josesimoes, 2019-02-19:
According to the message on the output there, basically you are referencing NuGet packages whose versions are not supported in the firmware that's running on your device.

The message content is trying to help you by describing the versions mismatches. 
If you pull the device's capabilities it has a list of the various assemblies it supports along with the versions. You have to reference the corresponding NuGets.

josesimoes, 2019-02-21:
Closing as there is no further input from the developer.

closed 2019-02-21 by josesimoes