nanoframework/Home #460: GPIO ValueChanged Events only fire for last opened pin

LoyalServant, 2019-02-26:
**nanoFramework area:** C# code

**VS version:** VS2017 

**VS extension version:** 1.0.3.60

**Target:**  ST Nucleo64 F411RE MB1136 C-04

**Firmware image version:** 1.1.47.0

Worked before? If so, with which nanoFramework image version: 
Unknown.

### Detailed repro steps so we can see the same problem

Attached application example exhibits issue. If the order that 'PhaseA' and 'PhaseB' are opened is changed the corresponding event will work. Only the first pin opened will have it's assigned event fire.

[Program.zip](https://github.com/nanoframework/Home/files/2904657/Program.zip)

LoyalServant, 2019-03-06:
New information... version 1.1.40.0 images with Windows.Devices.Gpio version 1.0.4 works.

josesimoes, 2019-03-06:
@LoyalServant the above is before or after your message on Discord?
Can we consider that this is working or you'll be sending a PR with a fix/improvement?

LoyalServant, 2019-03-06:
Unfortunately this is more or less over my head. There is some other complex interaction here that I don't understand since code written the same way in the .NET framework has no issue.
The regression is due to the made change here:
https://github.com/nanoframework/lib-Windows.Devices.Gpio/pull/72

Something more sinister is going on with the garbage collector since it eats the events if you call it and possibly a memory leak on top of all of this. If a pin event is fired a small number of times rapidly (less than 100) you crash with an out of memory:

Failed allocation for 17 blocks, 204 bytes

    #### Exception System.OutOfMemoryException - CLR_E_OUT_OF_MEMORY (4) ####
    #### Message: 
    #### nanoFramework.Runtime.Events.EventSink::GetEvent [IP: 0010] ####
    #### nanoFramework.Runtime.Events.EventSink::EventDispatchCallback [IP: 000e] ####
Failed allocation for 5 blocks, 60 bytes

Exception thrown: 'System.OutOfMemoryException' in nanoFramework.Runtime.Events.dll
Failed allocation for 5 blocks, 60 bytes

An unhandled exception of type 'System.OutOfMemoryException' occurred in nanoFramework.Runtime.Events.dll

josesimoes, 2019-03-06:
Following your last hint I've briefly looked at the code and I have an idea of what's causing the missed events.

On the memory exception that's to be expected on device with "low" memory. Each event causes a new object to be created. Eventually this will consume all the memory...

closed 2019-03-07 by josesimoes