nanoframework/Home #462: Network config hangs for NUCLEO 144 - F746ZG board

piwi1263, 2019-03-09:
### Details about Problem

**nanoFramework area:** Visual Studio extension | Hardware/target board | Community targets 

**VS version:** 15.9.8

**VS extension version:** v1.0.3.66

**Target:** ST_NUCLEO144_F746ZG

**Firmware image version:** 1.1.47.0

**Device capabilities output:** 
`** nanoFramework extension v1.0.3.66 loaded **

Add a short review or rate the VS extension: https://marketplace.visualstudio.com/items?itemName=vs-publisher-1470366.nanoFrameworkVS2017Extension
Star our GitHub repos: https://github.com/nanoframework/Home
Follow our YouTube channel: https://www.youtube.com/c/nanoFramework
Follow us on Twitter: https://twitter.com/nanoframework
Join our Hackster.io platform: https://www.hackster.io/nanoframework
Join our Discord community: https://discord.gg/gCyBu8T
Report issues: https://github.com/nanoframework/Home/issues
GitHub repo: https://github.com/nanoframework/Home
2019-03-09 02:50:20Z [NanoDevices: dropping black listed device \\?\ACPI#PNP0501#0#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:20Z [NanoDevices: new device arrival \\?\USB#VID_0483&PID_374B&MI_02#8&15ed06e&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73} ]
2019-03-09 02:50:20Z [NanoDevices: candidate nano device \\?\USB#VID_0483&PID_374B&MI_02#8&15ed06e&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:20Z [NanoDevices: new device arrival \\?\USB#VID_0483&PID_5740#NANO_3776316B323#{86e0d1e0-8089-11d0-9ce4-08003e301f73} ]
2019-03-09 02:50:20Z [NanoDevices: candidate nano device \\?\USB#VID_0483&PID_5740#NANO_3776316B323#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:20Z [NanoDevices: checking device \\?\USB#VID_0483&PID_374B&MI_02#8&15ed06e&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:22Z [NanoDevices: quitting device \\?\USB#VID_0483&PID_374B&MI_02#8&15ed06e&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:22Z [NanoDevices: checking device \\?\USB#VID_0483&PID_5740#NANO_3776316B323#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:22Z [NanoDevices: valid device NANO_3776316B323 @ COM20 \\?\USB#VID_0483&PID_5740#NANO_3776316B323#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-03-09 02:50:22Z [NanoDevices: Serial device enumeration completed. Found 1 devices]
2019-03-09 02:51:45Z [NanoDevices: open device \\?\USB#VID_0483&PID_5740#NANO_3776316B323#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
NANO_3776316B323 @ COM20 is active running nanoCLR

System Information
HAL build info: 1.1.47.0, nanoCLR running @ ST_NUCLEO144_F746ZG built with ChibiOS v18.2.2.2

Image build @ Feb 20 2019 GNU ARM GCC v7.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.1.47.0, nanoCLR running @ ST_NUCLEO144_F746ZG built with ChibiOS v18.2.2.2

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v1.1.1.7, checksum 0x7CCFBEC1
  nanoFramework.Runtime.Native v1.0.4.4, checksum 0xFE14C7A7
  nanoFramework.Hardware.Stm32 v1.0.4.3, checksum 0x024670A2
  nanoFramework.Devices.OneWire v1.0.3.3, checksum 0x4BFAF309
  nanoFramework.Networking.Sntp v1.0.4.3, checksum 0x733B4551
  nanoFramework.Runtime.Events v1.0.4.3, checksum 0xBFF88292
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v1.0.4.3, checksum 0x4BDCF00F
  System.Net v1.0.4.3, checksum 0x001C1FB9
  Windows.Devices.Adc v1.1.3.1, checksum 0xE5D11571
  Windows.Devices.Gpio v1.0.5.1, checksum 0xB1F30A6A
  Windows.Devices.I2c v1.1.3.1, checksum 0xA44C698B
  Windows.Devices.Pwm v1.1.3.1, checksum 0x75280B26
  Windows.Devices.SerialCommunication v1.0.6.1, checksum 0xA8A5A0E0
  Windows.Devices.Spi v1.1.3.0, checksum 0x083DBB79

--------------------------------
::        Memory Map          ::
--------------------------------
  Type     Start       Size
--------------------------------
  RAM   0x200000c0  0x00040000
  FLASH 0x08000000  0x00100000

---------------------------------------------------------------
::                   Flash Sector Map                        ::
---------------------------------------------------------------
  Region     Start      Blocks   Bytes/Block    Usage
---------------------------------------------------------------
      0    0x08000000       1      0x008000     nanoBooter
      1    0x08008000       1      0x008000     Configuration
      2    0x08010000       2      0x008000     nanoCLR
      3    0x08020000       1      0x020000     nanoCLR
      4    0x08040000       1      0x040000     nanoCLR
      5    0x08080000       2      0x040000     Deployment

---------------------------------------------------
::              Storage Usage Map                ::
---------------------------------------------------
  Start        Size (kB)           Usage
---------------------------------------------------
  0x08000000    0x008000 (32kB)     nanoBooter
  0x08008000    0x008000 (32kB)     Configuration
  0x08010000    0x070000 (448kB)    nanoCLR
  0x08080000    0x080000 (512kB)    Deployment

Deployment Map
Empty

`

Worked before? If so, with which nanoFramework image version: Unknown

### Detailed repro steps so we can see the same problem

1. Update board to FW version as above

2. Start VS

3. Select board in device explorer

4. Check & verify nanoCLR is running

5. Click on Network configuration button in device explorer

6. Select Ethernet interface type, set MAC address and click OK. 

7. Update text in status bar stays forever and VS seems to hang 

### Other suggested things
N/A

### Sample Project (if applicable)
N/A

### Screenshot (if applicable/relevant)
N/A

josesimoes, 2019-03-22:
Based on my recent incursion on this board, I have an idea on what could be happening.
This seems to be caused by a short CRT heap. At some point the code need to allocate 32k (that's the config block size) and there isn't enough heap for that on this target.

This probaly needs to be addressed by rewriting the code that manages this

piwi1263, 2019-03-23:
After some changes in the memory segment use in the CLR folder of the STM32F746ZG community board there is some success to having IP via DHCP and pinging the target board. Further stability tests are needed of course. Didn't work yet in a debug session only after it has been stopped.

piwi1263, 2019-03-25:
The workings have been confirmed and can be closed.

closed 2019-03-25 by piwi1263