nanoframework/Home #463: Problem with ESP32 Dev. C SPI Write function

Dweaver309, 2019-03-10:
### Details about Problem
I am getting this error when writing to a SPI device using the ESP32 Dev. C board.

#### Exception System.ArgumentException - 0xfd000000 (1) ####
    #### Message: 
    #### Windows.Devices.Spi.SpiDevice::NativeTransfer [IP: 0000] ####
    #### Windows.Devices.Spi.SpiDevice::Write [IP: 0007] ####
    #### NF.SPI.ST7735.Driver.ST7735::WriteData [IP: 0012] ####
    #### NF.SPI.ST7735.Driver.ST7735::.ctor [IP: 039c] ####
    #### NF.SPI.ST7735.Driver.Program::Main [IP: 001c] ####
Exception thrown: 'System.ArgumentException' in Windows.Devices.Spi.dll

I have tried many different pins and combinations and get the same error when 

I call the _spi.Write() function.

**nanoFramework area:** (C# code | Visual Studio extension | Hardware/target board | Nuget packages | Community targets | Community contributions | Documentation )

**VS version:** 
15.9.7

**VS extension version:** 
1.0.3.42

**Target:** 
ESP32 Dev. c board

**Firmware image version:** 
739

**SPI:** 
nanoFramework.Windows.Devices.Spi.1.1.3-preview-005

Worked before? If so, with which nanoFramework image version: 
Works with STM type boards

Code:
https://cdn.discordapp.com/attachments/481782754674212867/554086377889726467/Esp32nF.zip

Screen shot:
https://cdn.discordapp.com/attachments/481782754674212867/554086014658805761/unknown.png

### Detailed repro steps so we can see the same problem

1. Compile code you will see the Screen shot above

AdrianSoundy, 2019-03-12:
This is caused when trying to write more then 4094 bytes in one write operation.
Currently the default for ESP32 max buffer is 4094 which is set up when Bus is configured.

This could be set up with a different values based on a new connection settings.

But for the moment its best to limit the write size under this limit.

closed 2019-03-12 by AdrianSoundy

Dweaver309, 2019-03-12:
Thanks Adrian!