nanoframework/Home #464: System.Math Not Implemented Exception on Various Functions

gligorov, 2019-03-14:
### Details about Problem

When using functions in the Math library, for example Math.Pow(), it throws Not Implemented exception.  The library is included in the build and the app builds fine. At run time throws the exception when the functions are used.

I tested it with System.Math v1.0.4.3, checksum 0x4BDCF00F at the moment.

Latest F4 Image but any board/image will do it.

### Detailed repro steps so we can see the same problem

1.  Just build any app, and attempt to use  Math.Pow() function you will get the exception at run time.

josesimoes, 2019-03-18:
I've tried the following code and it's working as expected

```csharp
namespace math_test
{
    public class Program
    {
        const float angle90degrees = (float)(Math.PI) * 90.0f / 180.0f;

        public static void Main()
        {
            Console.WriteLine($"sin(90): { Math.Sin(angle90degrees) }");

            Console.WriteLine($"cos(90): { Math.Cos(angle90degrees) }");

            Console.WriteLine($"2^2: { Math.Pow(2, 2) }");

            Console.WriteLine($"log(e): { Math.Log((float)Math.E) }");

            Console.WriteLine($"log10(1): { Math.Log10(1) }");

            Thread.Sleep(Timeout.Infinite);
        }
    }
}
```

output:

```
sin(90): 1
cos(90): -0.000000043000000000
2^2: 4
log(e): 1
log10(1): 0
```

[math-test.zip](https://github.com/nanoframework/Home/files/2979321/math-test.zip)

closed 2019-03-18 by josesimoes

gligorov, 2019-03-18:
This works because it uses only floats. It fails if using double in any method.

gligorov, 2019-03-18:
public class Program
    {
        const double angle90degrees = Math.PI * 90.0 / 180.0;

        public static void Main()
        {
            double num1 = 2, num2 = 2;

            Console.WriteLine($"sin(90): { Math.Sin(angle90degrees) }");

            Console.WriteLine($"cos(90): { Math.Cos(angle90degrees) }");

            Console.WriteLine($"2^2: { Math.Pow(num1, num2) }");

            Console.WriteLine($"log(e): { Math.Log((float)Math.E) }");

            Console.WriteLine($"log10(1): { Math.Log10(1) }");

            Thread.Sleep(Timeout.Infinite);
        }
    }

gligorov, 2019-03-18:
It builds without problems as written above, but at runtime fails on the first operation. So if we need to only use float is ok, just did not know about it.

josesimoes, 2019-03-18:
That is correct. The default for the Math API is support for floats. That was changed quite sometime ago. Pulling in support for double FP calculation adds a LOT of code from the CRT libs.
For most applications that extra precision is not required at all.

Nevertheless it's available as a build option and can be used, if required. :wink:

gligorov, 2019-03-18:
That clarifies it, will just use float for now, tks.