nanoframework/Home #466: Visual Studio 2017 extension package did not load correctly

DeHaanSoftware, 2019-03-20:
After installing the vsix and restarting Visual Studio 2017 I get the message "The nanoFramework package did not load correctly". See attached screen shot
[image]

Visual Studio 2017 Community 2017 version 15.9.9 
Tried on both my Windows 10 laptop and my Windows 7 64-bit desktop

josesimoes, 2019-03-20:
@DeHaanSoftware could you please attach VS activity log (pointed in the error message).

BTW, the extension won't work on Windows 7 because of the USB UWP API that the debugger library uses.

DeHaanSoftware, 2019-03-20:
Thaks for the comment. So Windows 7 can't be used. Windows 10 works fine now after updating to the latest VS2017 version.

closed 2019-03-20 by DeHaanSoftware

tatianilima, 2023-04-09:
I experienced the same error with Windows 11 and Visual Studio 2022. 
[image]

Does anyone know how to solve it?

Ellerbach, 2023-04-09:
The latest version has an issue, please use the one before.