nanoframework/Home #470: Error adding file resources in VS

dlkay0, 2019-03-29:
### Details about Problem
The nanoFramework ResX editor behaves incorrectly when adding resources.  The normal ResX editor handles these cases fine.

**nanoFramework area:** Visual Studio extension
**VS version:** 15.9.10
**VS extension version:** 1.0.3.92

### Detailed repro steps so we can see the same problem

Scenario A - VS Exception
1. Place a binary file in your _project's_ folder
2. Add a nanoFramework Resource file to your project
3. Click Add Resource --> Add Existing File...
4. Choose the file you placed into your project's folder and click OK
5. Observe Exception and selected file is added to your project as an embedded resource but not in the ResX file.
[image]

Scenario B - Destination File Exists
1. Place a binary file in your _solution's_ folder
2. Add a nanoFramework Resource file to your project
3. Click Add Resource --> Add Existing File...
4. Choose the file you placed into your project's folder and click OK
5. Observe your file is embedded under a new Resources folder
6. Observe Dialog
[image]
7.  Answer "Yes" and get the Exception again.  In either case, the ResX file is not updated.

Scenario C - Resource already added to project
1. Before adding the resource using the resource editor...
  a. Right-click your project and select Add --> Existing Item...
  b. Choose your binary file and click Add
  c. Right-click the file in Solution Explorer and select Properties
  d. Change Item Type to Embedded Resource and click OK
2. Add the same file using ResX editor and it succeeds.

josesimoes, 2019-03-29:
@dlkay0 a 🏅  for your issue reports! 
It's a delight to see the level of detail and cleanness! Brilliant. 👏  😄

josesimoes, 2019-04-08:
This exception is coming from VS project system. Apparently something related with the resource icon handling.
An issue has been opened in VS developer community [here](https://developercommunity.visualstudio.com/content/problem/522682/exception-when-adding-a-resource-file-to-nanoframe.html).

Please up vote it to move it up in their priority list. 😉

networkfusion, 2019-07-10:
Just to add, if you manually modify the resx e.g. 
```
  <data name="CAroot" type="System.Resources.ResXFileRef, System.Windows.Forms">
    <value>Resources\CAroot.pem;System.Byte[], mscorlib, Version=1.2.6.0, Culture=neutral, PublicKeyToken=c07d481e9758c731</value>
  </data>
```

you cannot compile due to:
[image]

josesimoes, 2019-07-10:
You aren't supposed to manually edit anything on those files, otherwise you'll risk breaking the build.

Something worth noting about that `CodeDom` exception: that's caused because the tool associated with it is not properly set.

For nanoFramework resources you have to set the custom tool to `nFResXFileCodeGenerator` like this:

[image]

This is automatically set if you add a resource file from VS add new dialog and choose the resource under nanoFramework tree. If you add it manually or from other language, it's required to manually tweak the tool.

networkfusion, 2019-07-10:
Interesting, I used the VS Add File tool to add a Resource File (nanoFramework) and the custom tool was set as something else. I have now changed it to what has been suggested and will see what happens.

josesimoes, 2019-07-17:
Issue in VS community has been update to "Under Consideration".

It would be great if:

## more people upvote the issue [there](https://developercommunity.visualstudio.com/content/problem/522682/exception-when-adding-a-resource-file-to-nanoframe.html)

networkfusion, 2019-07-17:
I would suggest an ATeveryone on discord for this...

networkfusion, 2019-08-13:
from the VS feedback, a possible solution is:

Based on the rules files you're using, it seems like you've taken from the examples in the Project System GitHub. Unfortunately, those are stale because our SDK is not adding the appropriate properties when generating these files. We'll look into this error and fix the sample as well as the SDK.
For now, please use the .NET Core rules files as reference. This particular error is being caused by None items, for which you can find the template here. You should also take a look at the other files and add the necessary properties to yours based on that (such as Item Type, custom tool, etc).

josesimoes, 2019-08-15:
Need the blocked label, otherwise stale bot picks this has stale 😉

closed 2019-08-20 by josesimoes