nanoframework/Home #471: Update SslProtocols enum

josesimoes, 2019-03-29:
- [ ] Remove SSLv3 (it's deprecated, outdated and unused) no need keeping it around just to waste flash bytes.
- [ ] Remove Default. The full .NET framework as stopped using this a long time ago. The "default" is now None as described in the [documentation](https://docs.microsoft.com/en-us/dotnet/api/system.security.authentication.sslprotocols?view=netframework-4.7).
- [ ] Replace wherever we have on our code base `SslProtocols.Default` with `SslProtocols.None` to follow the full .NET.

RoSchmiSec, 2019-04-04:
After thinking about these suggestions I'd suggest to keep Default but set it to Tls1.2. If everywhere in the code SslProtocols.Default is replaced by SsslProtocol.Non all existing applications would be broken. Additionally I suggest to implement a method to change the used Ssl protocol from the application side.

josesimoes, 2019-04-04:
When using `SslStream` we can already choose which SSL protocol to use.

I guess you are referring to the other classes that rely on `SslProtocols.Default` like HttpWebRequest and such. The draw back is that there are still servers that are on TLS v1.1 and even v1.0. These haven't been deprecated yet and there are only recomendation for people making the move. See [here](https://en.wikipedia.org/wiki/Transport_Layer_Security).
So, enforcing this would be a bit pushy from us...

I would rather follow your other suggestion of exposing a property allow the calling apps to set the SSL protocols. That's exactly what the .NET framework did by deprecating HttpWebRequest and friends replacing it with HttpClient that exposes this through it's HttpClientHandler.

Until we have our own HttpClient, I believe the best was to deal with this is add that property in the classes where that makes sense.

josesimoes, 2019-04-05:
@RoSchmiSec can you please take care of this one? 😉

closed 2019-06-04 by josesimoes