nanoframework/Home #474: Issues with HttpListener & Socket

weixiongmei, 2019-04-06:
Processing HttpListener.GetContext() or Socket.Accept() with HttpListenerContext.Response or Socket.Receive() in a single thread, it will throws socket exception randomly, this problem can be repeated by the HTTP sample code, once the code is running on the ESP32 in debug mode, just open the website in the browser & refresh the page couple times, the exception will occur. This problem can be solve by creating the new thread to process HttpListenerContext.Response or Socket.Receive(), but this multi thread solution solved the exception problem, but cause another problem. In the multi thread solution, the HttpListener will cause the TTFB(Time To First Byte) delay between 800ms to 3000ms, most of the time, each request caused a 1200ms delay. The Socket in the multi thread solution can accept all the connections, but lost data when calling Socket.Receive(). The problem seems like the process of the Socket class behind the scene is processing too slow, so the data over flows and causing these problems.... Thankyou~~

josesimoes, 2019-04-06:
Interesting problem that you're describing...
It seems that you've already made some investigation and possible solutions. Could you please attach a project were the issue and the possible fixes are? It would be very helpful.

weixiongmei, 2019-04-06:
[image]
Single thread cause socket exception if more than one request in a short period of time. (Not sure the time frame, but refreshing the website in the Chrome will cause the problem, because the Chrome all ways request the favicon.ico right after the first request.)

weixiongmei, 2019-04-06:
[image]

Multi thread the permanently solve the socket exception problem, but cause the TTFB delays, most of the request would delays 1000ms - 2000ms.

weixiongmei, 2019-04-06:
[image]

The socket exception can be solved by using the multi thread solution for the Socket class too, but the multi thread solution will causing data lost that sent from the client, you can do the same test in the Chrome, the second request for the favicon.ico can be accepted, but not receiving any data from client.

weixiongmei, 2019-04-06:
[Wifi.zip](https://github.com/nanoframework/Home/files/3050727/Wifi.zip)

weixiongmei, 2019-04-06:
[Wifi.zip](https://github.com/nanoframework/Home/files/3050728/Wifi.zip)
Here's the project that including all the codes for HttpListener & Socket

josesimoes, 2019-04-06:
Thank you very much for sending the test coe and for the detailed explanations! 👍 

From the looks of it this is most likely related with the way the execution engine handles the timeouts and continuations internally.
Note that the sample was meant to be a plain simple "hello world" not a web server by any stretch. That would require a thread per request, gracefully closing sockets, disposing, etc.

Have you looked into code for old .NETMF web servers out there, to see how this problem was handled?
A couple of examples from a quick search (the 1st one I've used it in the past)
https://weblogs.asp.net/mschwarz/my-own-small-net-micro-framework-web-server
https://github.com/Rinsen/WebServer

weixiongmei, 2019-04-06:
> Thank you very much for sending the test coe and for the detailed explanations! 👍
>
> From the looks of it this is most likely related with the way the execution engine handles the timeouts and [...]

Hi, the nanoframework can't even handle the hello world http sample, when there's is more than one request sent to the server, even just two simple http requests. For instance, using the Chrome browser, open a web page, it always sent to requests to the server, one is for the URL that's entered, one is for favicon.ico, so the nanoframework always throw the socket exception(i'm assuming it's because the second request arrived before the finishing the first request).... I also tested it with the Edge browser, the Edge only sent one request when opening a web page, so the nanoframework can handles it with no problem. The problem now is, even just one connection, and sending more than one request in a short period of time(lets say two requests within 20ms), the nanoframework will throws the socket exception...

josesimoes, 2019-04-08:
The HttpListener sample has been removed as it has the potential to mislead developers that it can perform as web server. It can not.

A web server requires that a request is handled on it's own thread, served and grecefully closed. All this without blocking incomming requests.

Some pointers to projects that have implemented Web Servers in the past for .NETMF have been provided above. Others exist.

In the future we might provide our own _nano_ webserver. A task for this has been added to the Networking project [here](https://github.com/orgs/nanoframework/projects/6#card-19955432). Anyone willing to tackle it is welcome.
Initial contacts with a couple of those projects have been made.
The initial stage of such component shouldn't require much work apart from replacing namespaces and a tweak in the code, here and there.

closed 2019-04-08 by josesimoes