nanoframework/Home #475: Class library's can not be referenced

Dweaver309, 2019-04-16:
### Details about Problem

**VS version Microsoft Visual Studio Community 2019 RC Version 16.0.0** 

**VS extension version 1.0.0.105** 

**Firmware image version 1.0.4-preview.404** 

### Detailed repro steps so we can see the same problem

1. Built the class library

2. Create a new application

3. Reference the .dll created  

4. The reference  will not be available

### Sample Projects

[NFClassLibrary2.zip](https://github.com/nanoframework/Home/files/3085663/NFClassLibrary2.zip)
[TestClsLib.zip](https://github.com/nanoframework/Home/files/3085664/TestClsLib.zip)

 

...

Dweaver309, 2019-04-18:
I know nothing about vsix extensions.  However I looked at in earlier NF extension and the newest one. I saw that this version had an extra directory named CSharp.ResourceTemplate.  I hope this helps.
[image]

josesimoes, 2019-04-18:
I'm afraid that's totally unrelated...
That's the new template for managed resources that was added a few weeks ago.

Dweaver309, 2019-04-18:
Thanks that is what I thought.

Dweaver309, 2019-04-22:
The class library's that are referenced can't be viewed in the object browser.

josesimoes, 2019-04-30:
Reproduced. It works on VS2017, but not on VS2019.
Something must be missing related with a configuration that as to be requested for nF project type.

As a side note: VS Object explorer can't load the referenced libraries when double clicking, not navigate to the objects using F12.

josesimoes, 2019-05-06:
The root cause for this seems to be with VS Intellisense engine.

Reported the issue on the developer community [here](https://developercommunity.visualstudio.com/content/problem/558572/intellisense-issue-with-referenced-assembly.html). Waiting for feedback.

josesimoes, 2019-06-17:
Probably solved with microsoft/msbuild#4348.
Need to wait for VS 2019 v16.2 for confirmation.

networkfusion, 2019-07-17:
Now available in the preview release. Waiting for the official release before this can be closed.

josesimoes, 2019-07-17:
@networkfusion you have the preview installed and it's working now?

networkfusion, 2019-07-17:
@josesimoes oops, no, but Preview 4 has just been released which closes 4348?!

josesimoes, 2019-08-01:
Move from VS Community to GitHub dotnet/project-system#5239

networkfusion, 2019-09-27:
not sure if this helps yet... but adding for reference: https://referencesource.microsoft.com/#MSBuildFiles/C/ProgramFiles(x86)/MSBuild/14.0/bin_/amd64/Microsoft.CSharp.Core.targets

closed 2019-10-17 by josesimoes