nanoframework/Home #476: Bluetooth support

homerokzam, 2019-04-18:
Are you planning to support Bluetooth?

Thanks

homerokzam, 2019-06-04:
?

networkfusion, 2019-06-07:
Yes, it is something on the list, but not available out of the box yet. However, if you have a serial Bluetooth adapter, it is possible to write your own interactions in managed code (C#).

If you would like to discuss further, please use our discord channels. Although I will leave this request open for when it is implemented.

homerokzam, 2019-07-30:
I developed an app (ios and android) to communicate with devices that have BLE. I made it work on ESP32 developing in C, but I don't know how to start writing the firmware in C # (using nanoframework).

I'm a software guy, and I have some difficulty understanding how the hardware works.

josesimoes, 2019-07-31:
@homerokzam nanoFramework does not have support for BLE on ESP32. Because there isn't an API we can't help you on porting your code to C#.

Unless I'm missunderstanding your message and you are offering to start develop BLE support in ESP32 for nanoFramework. Please clarify.

For conversations suggest that you join our [Discord community](https://discord.gg/gCyBu8T). Better suited for conversations! 😉

homerokzam, 2019-07-31:
At first I would like to use an api with BLE. I can help develop, but I have no hardware experience. This is my first hardware project and my background is in software.

I will access the Discord community.

Thanks.

Hassmann, 2020-04-21:
@homerokzam Are you still interested in this? I want to start adding BLE to ESP32.

homerokzam, 2020-05-04:
Yes, I'm interested. But I'm a software guy, not a hardware guy.

Thanks.

josesimoes, 2020-05-21:
ℹ️ @Hassmann is working on this.

josesimoes, 2020-07-20:
In progress. Correct @Hassmann ?

Hassmann, 2020-08-03:
This is unfortunately not in progress anymore. I am stuck here.

The reasons are:

1. The espressif native BT libraries are way too large to include them into a standard image.
2. They are even too large for including them into a custom firmware without serious cropping.
3. NimBLE would fit, but it relies on ESP-IDF 4, currently not supported by nF

The state of the work done is in:

https://github.com/Hassmann/lib-nanoFramework.Hardware.Esp32.Bluetooth
https://github.com/Hassmann/nf-interpreter

If anyone wants to take over, I am happy to assist.

danielmeza, 2020-10-07:
> This is unfortunately not in progress anymore. I am stuck here.
>
> The reasons are:
>
> 1. The espressif native BT libraries are way too large to include them into a standard image.
> 2. They are even too [...]

So the answer is to support ESP-IDF 4? What about to use the Arduino library implementation for ESP?

josesimoes, 2020-10-07:
@danielmeza the Arduino library is build on top of the IDF, which would bring us to the same place.
I'm afraid that this needs to wait until our build is migrated to IDF-4...

danielmeza, 2020-10-07:
@josesimoes I guest that too, can you point us to that task in order to help with the migration to IDF-4? Perhaps a check list with the pending's to reach IDF-4 migration?

josesimoes, 2020-10-07:
@danielmeza glad that you've asked! 😄 

Migrating the IDF involves several steps. Here's a general overview of the process.
- Recompiling the libraries see [here](https://docs.nanoframework.net/content/esp32/build_ESP32_libraries.html)
- Updating the various references to the new IDF version in the CMake files and PS scripts (in nf-interpreter repo)
- Adjust calls to the IDF functions/API, if needed.

For details suggest the #native-esp32 channel on Discord.
Also know that @AdrianSoundy is **the** ESP32 master.

AdrianSoundy, 2021-07-22:
The latest nanoframework IDF update to v3.3.5 now includes the bluetooth component ready for a managed library to be built. This is based on Nimble.
There were problems including bluetooth due to size  especially as a lot of iram was used by the PSRAM fixes. So currently the is a bluetooth version without psram for version 1 ESP32. For Version 3 ESP32 it was not problem as psram fixes not required.

AdrianSoundy, 2021-07-22:
For managed library we can either base the library from the UWP Windows.devices.Bluetooth or continue with the work done by @Hassmann. 
There are a lots of classes to implement in the UWP library, maybe we should create our own based on UWP but called System.Device.Bluetooth.

Ellerbach, 2021-07-23:
> For managed library we can either base the library from the UWP Windows.devices.Bluetooth or continue with the work done by @Hassmann.
> There are a lots of classes to implement in the UWP library, maybe we should create our own based on UWP but called System.Device.Bluetooth.

I would be pragmatic and use Syste.Device.Bluetooth, move any good work done previously and maybe just the first subset of necessary classes to start with.

AdrianSoundy, 2021-08-09:
Started implementing Gatt classes but hit a blocker when looking to implement an event to managed code for gatt property changes read/write. The callbacks for nimble expects you to respond directly in callback for property changes not allowing event to b reflected to managed code

romarro, 2021-09-21:
@AdrianSoundy have you had any success with the Bluetooth library?

AdrianSoundy, 2021-09-21:
Still working on it. 
Hitting memory issues on ESP32.

josemalm32, 2021-10-17:
Hi @AdrianSoundy any news?

Thanks 🥇

josesimoes, 2021-12-03:
First preview release for the library has been published. See [nanoFramework.Device.Bluetooth](https://github.com/nanoframework/nanoFramework.Device.Bluetooth)

Closing.

closed 2021-12-03 by josesimoes