nanoframework/Home #484: Invalid type after assigning an array to the object

MateuszKlatecki, 2019-06-13:
## Details about Problem

**nanoFramework area:** ( mscorelib or compiler)

## Description

After assign array (no matter what type) to object its type cannot be determined.

Sample code:
```cs
using System;

namespace nfTest
{
    public class Program
    { 
        class MyType
        {

        }
        static void Main(string[] args)
        {
            var o = new int[] { 1234 };

            if (o is byte)
                Console.WriteLine("Byte");
            else
                Console.WriteLine("Not Byte");

            object obj = o;

            if (obj is byte)
                Console.WriteLine("Byte");

            if (obj is int)
                Console.WriteLine("Int");

            if (obj is String)
                Console.WriteLine("String");

            if (obj is Program.MyType)
                Console.WriteLine("MyType");

        }
    }
}
```
Console output
```console
Not Byte
--=== After assign ===--
Byte
Int
String
MyType
```

Interestingly, if it is not an array, the type after the assignment is recognized correctly

## Expected behaviour

After assigning the table to the object, it is still possible to recognize the type

josesimoes, 2020-02-12:
@MateuszKlatecki when you have the opportunity can you please run this test again? It might have been fixed with nanoframework/nf-interpreter#1547...

MateuszKlatecki, 2020-02-13:
Unfortunately, this change does not fix my case.

closed 2020-04-27 by josesimoes