nanoframework/Home #495: Update URL for documentation on NuGet readme

josesimoes, 2019-06-24:
The documentation website has changed its organization. The docs home is now:
https://docs.nanoframework.net/content/intro.html

Need to update the readme for the NuGet packages on all class lib repos.

kurtnelle, 2019-07-11:
Grabs

josesimoes, 2019-07-11:
@kurtnelle suggest that you edit the file right on GitHub, then create a branch when asked.
It's the easier way to handle these small changes without having to go all the trouble of creating local repos, branches and submit PRs. And still meeting all the ceremony that this requires. 😉 

Of course this is just a suggestion. Feel free to handle it any way that works for you!

kurtnelle, 2019-07-12:
Ok, I was confused for a bit there. There is not intro.html page under content. So the URL is just: https://docs.nanoframework.net/content/

Is that ok?

josesimoes, 2019-07-12:
You are correct.
Actually https://docs.nanoframework.net just works.

kurtnelle, 2019-07-14:
All projects updated.

josesimoes, 2019-07-15:
Readme updated on all class lib repos. Closing.

closed 2019-07-15 by josesimoes