nanoframework/Home #497: spi.write on ESP32 writes erronous data to the bus

kurtnelle, 2019-06-30:
public static void Main()
	{
		
		var spi = SpiDevice.FromId("SPI1", new SpiConnectionSettings(22));

		while (true)
		{
			spi.Write(new byte[] { 0x0 });
			Thread.Sleep(1000);
		}
	}

When I run this code on an ESP32, and monitor the output on my scope, I see a wall of data being sent (whereas the pin should just stay low). If I try and set the spi speed in the Connection Settings, the function spi.Write never returns, and just keeps toggling that output pin. The only way to get the device to respond after that is to erase the program and reset.

josesimoes, 2019-06-30:
Can you please post a screenshot of the scope output?

Looking at the above code a continuous ouptut of 0x00 on the SPI bus is expected.

When you change the SPI bus speed to an erroneous value that can screw it. You don't mention to what value you're setting it... 

On the device becoming irresponsible that can happen too. The app is continously writing to the bus with 1sec interval to breath. It could be hard for the debugger comms to pick-up the incoming requests...

PS: please respect the issues template next time. 🙂

kurtnelle, 2019-07-01:
[image]
This is a trace of the start of the data

[image]
This seemingly random pattern goes on for a while. The slower I set the clock frequency, the longer the pattern runs for. I had to change the frequency because I was getting a distorted signal on the scope.

````   
public class Program
    {
        public static void Main()
        {
            
            var spi = SpiDevice.FromId("SPI1", new SpiConnectionSettings(22) { ClockFrequency = 1 * 1000 * 1000 });

            while (true)
            {
                spi.Write(new byte[] { 0x0 });
                Thread.Sleep(10000);
            }
        }
    }
````

kurtnelle, 2019-07-01:
Sorry, context. Purple is CS yellow is MOSI and blue is CLK

josesimoes, 2019-07-01:
Agreed: that doesn't feel right.

AdrianSoundy, 2019-07-01:
The problem is caused by not settings the DataBitLength property in the SpiConnectionSettings

var spi = SpiDevice.FromId("SPI1", new SpiConnectionSettings(22) { ClockFrequency = 1 * 1000 * 1000,  DataBitLength = 8  });

I'll add a change to make sure it defaults to 8 if not specified instead of 0

kurtnelle, 2019-07-02:
The proposed solution works!

[image]

closed 2019-07-02 by kurtnelle