nanoframework/Home #498: I2C does nut have an output when I2CDevice is statically initialized

rayperkins, 2019-07-02:
## Details about Problem

**nanoFramework area:** (C# code )

**VS version:** 16.1.2

**VS extension version:**  1.1.0.121

**Target:**  ESP32

**Firmware image version:** 1.0.4.503

**Device capabilities output:**
  mscorlib v100.2.1.0, checksum 0x17723A6B
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.6.0, checksum 0xFF4537C1
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.5.0, checksum 0x9ACFB207
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.1.0, checksum 0x3AD3B5E6
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.0.0, checksum 0xF3FBAE36
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

## Description

Initializing the I2C device as a static member will only show I2C activity when the VS2019 debugging session is attached. When the device is reset after a debug session then the I2C bus has no activity on the pins.

Furthermore, I observed an occasional extra clock pulse, it seemed to be repeat roughly every second.

## Detailed repro steps so we can see the same problem

1. Open the attached project and configure the pin mappings to suit.

2. Run the project with the debug session active and observe the CLOCK pin output is correct.

3. Stop the debug session and restart the board. The CLOCK pin will not show any activity.

...

## Other suggested things
Perhaps there is some sort of race condition for the SetPinFunction to remap the I2C bus pins.

## Expected behaviour

The I2C should operate correctly, 
regardless of the VS debug session. 

### Sample Project
[NanoFramework.ESP32TestApp.zip](https://github.com/nanoframework/Home/files/3347834/NanoFramework.ESP32TestApp.zip)

## Screenshot
[image]

rayperkins, 2019-07-03:
I suspect the lack of I2C when running not from a debugger is caused by the pins not being remapped in the Main method until after the I2C Device has been configured. 

I think that the reason that it works when in debug is that when the program is deployed it will start running (and in this example it will remap the I2C pins in the Main method). 
The VS2019 debugger will attach after deployment and restart the program. Now the program starts for the second time and initializes the I2CDevice using the remapped pin config from the previous program execution.

When the board is power cycled the device will do the obvious by initializing the I2C pins to defaults and start up the C# program which creates an I2CDevice, using the default pins.

The TL;DR; is: this example misuses the I2CDevice by attempting to remap the pins *after* initializing I2C.

The extra bonus clock pulse is still a mystery. Perhaps the ESP-IDF driver is attempting a delayed repeated start? (No slave device was attached when scoping the pins for the attached clock image).

josesimoes, 2019-07-03:
I still haven't seen the code on your project but if you are initializing the pins statically that's the explanation.
When there is a debugger attached the static constructors are initialized before the app starts to run. This is because there could be breakpoints set in the the static constructors and we don't want to miss those. 

The remedy is to init the pins and I2C inside the app. 

As for the clock pulse I have no explanation to offer at this time... 😉

rayperkins, 2019-07-04:
Alrighty, I'll close the issue since the above comments explain the observed behavior as something that is expected.

The extra pulse can be raised as a new issue in the future, if it is indeed an issue.

closed 2019-07-04 by rayperkins