nanoframework/Home #499: System.Delegate allow to get InvocationList

MateuszKlatecki, 2019-07-02:
**nanoFramework area:** corlib

**Is your feature request related to a problem? Please describe.**
There is no way to get an array of delegates representing the invocation list of the current delegate.  The current version only allows to get the recently added delegate (Method getter) 

**Describe the solution you'd like**
I would like to have the getInvocationList method as in the standard .net returning an array of all delegates.
https://docs.microsoft.com/en-US/dotnet/api/system.delegate.getinvocationlist?view=netframework-4.8

**Example**

```cs
public delegate void TestEventHandler();

public class Program
{
    public static event TestEventHandler OnTest;
    public static void Main()
    {
        OnTest += Test1EventListener;
        OnTest += Test2EventListener;

        //return last added delegate
        var lastDelegate = OnTest.Method;

        //no OnTest.GetInvocationList() method which should return System.Delegate[2]
        //but not exists

        OnTest.Invoke(); //trigger both listeners meand data of all delegates is available

    }

    private static void Test1EventListener()
    {

    }

    private static void Test2EventListener()
    {

    }
}
```

josesimoes, 2019-07-02:
@MateuszKlatecki are you working on a test framework? 😄

MateuszKlatecki, 2019-07-02:
@josesimoes No, it's just a convergence of names :)
We need this functionality to be able to serialize delegates so that they can be saved to a file / send over the network.

closed 2019-07-10 by josesimoes