nanoframework/Home #501: Socket class unable to receive UDP datagram.  Error 10050.

dlkay0, 2019-07-10:
Please read the following information before posting the issue.

# **Before posting the issue**

If you need clarification on something, need help on a particular situation or want to start a discussion, instead of opening an issue it is best to start a conversation on one of our Discord channels.
Go there and select the channel that's most appropriate to the context so subject matter experts are most likely to answer promptly.
You can join our lively [Discord community](https://discordapp.com/invite/gCyBu8T)

:exclamation: Remove the content above here and fill out details below. :exclamation:

## Details about Problem

**Target:** ESP32

**Firmware image version:** develop Tue Jul 9 15:30:16 2019 +0100
+ minor changes to enable LAN8720 in Esp32_Ethernet_Lan8720.cpp

Worked before? Unknown

## Description
When attempting to receive a UDP datagram, nF throws SocketException.  Same code works in Full Framework on Windows.

## Detailed repro steps so we can see the same problem
1.  Open a Socket in UDP datagram mode
2.  Send a UDP packet to the device's IP address/port
3. Observe socket.Available > 0
4. Call socket.Receive or socket.ReceiveFrom
5. SocketException thrown with ErrorCode 10050

## Serial output
```Running...
Got IP: 10.216.139.136
    #### Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ####
    #### Message: 
    #### System.Net.Sockets.NativeSocket::recvfrom [IP: 0000] ####
    #### System.Net.Sockets.Socket::ReceiveFrom [IP: 0033] ####
    #### System.Net.Sockets.Socket::ReceiveFrom [IP: 0011] ####
    #### NFApp2_EthTest.Program::Main [IP: 0055] ####
Exception thrown: 'System.Net.Sockets.SocketException' in System.Net.dll
ErrorCode: 10050
System.Net.Sockets.SocketException: Exception was thrown: System.Net.Sockets.SocketException
```
## Other suggested things
I tried unicast and broadcast UDP packets.  I tried various overloads.

## Expected behaviour
UDP packet data received into buffer.

## Additional context

[nFProgram.cs.txt](https://github.com/nanoframework/Home/files/3379486/nFProgram.cs.txt)
[Sender.cs.txt](https://github.com/nanoframework/Home/files/3379491/Sender.cs.txt)

josesimoes, 2019-07-30:
Can't reproduce this.
Tested with your code above (both sender and receiver).

Output from ESP32-WROVER-KIT + sender console app 

[image]

Output from STM32F769I-DISCO + sender console app

[image]

josesimoes, 2019-07-30:
Notice your note above about 

> minor changes to enable LAN8720 in Esp32_Ethernet_Lan8720.cpp

Suspec that the reason for failing could be on those changes...

josesimoes, 2019-08-14:
No further developments or feedback from the author. Closing.

closed 2019-08-14 by josesimoes

Shraddha246, 2021-04-09:
Hello NF,
Nanoframework firmware image with Ethernet support for Olimex ESP32-POE board.
We have tried to update NFinterpreter and flashed it on Olimex ESP32-POE board.
We got IP address but we are getting socket exception while connecting to MQTT service. Is there any plan to create separate firamware for ESP32 with Ethernet support. Please let me know. Thanks.