nanoframework/Home #502: M2Mqtt cannot parse hostname

networkfusion, 2019-07-11:
#167  Details about Problem

**nanoFramework area:** (C# code | Nuget packages )

**VS version:**
VS2019 (latest)

**VS extension version:**
Latest

**Target:**
STM32F769I

**Firmware image version:**
Latest

## Description
When using the following code, an internal exception occurs:
```
string broker address = "<endpoint>.azure-devices.net"
MqttClient mqttc = new MqttClient(iotBrokerAddress, 8883, true, null, null, MqttSslProtocols.TLSv1_2);
```
This method worked fine in NetMF, so I expect it is a change in nF.

As a work around, the following can be used (although the method is marked as obsolete)
```
string broker address = "<endpoint>.azure-devices.net"
// get host entry
IPHostEntry hostEntry = Dns.GetHostEntry(iotBrokerAddress);
//parse by IP
MqttClient mqttc = new MqttClient(hostEntry.AddressList[0], 8883, true, null, null, MqttSslProtocols.TLSv1_2);
```

josesimoes, 2019-07-11:
Can't see `iotBrokerAddress` declaration on your 1st block of code.

networkfusion, 2019-07-11:
Sorry, coding in the issue leads to mistakes... Edited...

networkfusion, 2019-07-11:
The issue is with `MQTTNetworkChannel` line `86` (`                remoteIpAddress = IPAddress.Parse(remoteHostName);`)

josesimoes, 2019-07-12:
The exception you're seeing in VS output is expected. It's a empty catch block in `MqttNetworkChannel(...)` constructor when it's trying to use the host name "as is" testing if an IP address was supplied.

[image]

Because nanoFramework is really awesome this is not required anymore and the correct approach can be DNS_ing_ whatever is in the host name. If it's a name, it will be resolved. If it's an IP, it will be resolved directly as an IP address.

I'll submit a change with this improvement.

closed 2019-07-12 by josesimoes