nanoframework/Home #504: Add X509Certificate2 to System.Net API

josesimoes, 2019-07-16:


josesimoes, 2019-07-16:
In progress just needed a placeholder for related discussions and notes.

kurtnelle, 2019-07-16:
Does this mean we can connect to https sites easily?

josesimoes, 2019-07-16:
You can already connect to HTTPS sites now. There is support for `X509Certificate`.
Check the SSL sample pack. There is a sample connecting to a couple of HTTPS test servers and performing a simple HTTP get request.

This one is to address a different problem: dynamically load the certificate and private key, choose the encryption password and get the final certificate from combining those.

josesimoes, 2019-07-16:
* Discussion on how to get this done with .NET Core [here](https://github.com/dotnet/corefx/issues/14284)

* API wih this improvement on .NET Core [here](https://github.com/dotnet/corefx/issues/17892)

networkfusion, 2019-07-16:
To start with, as long as we can parse a pfx certificate in bytes, then we should be able to connect (as that already contains the private key).

The full framework supports the setting of the private key, but core does not (currently). I would suggest our implementation does allow for it...

networkfusion, 2019-07-16:
The 2 classes that I use with the full framework to help with converting the cert above what .net offers are

```
//Source from https://github.com/AlexanderPro/Utilities

using System;
using System.Security.Cryptography;
using System.Security.Cryptography.X509Certificates;
using System.IO;
using Utilities.Extensions;

namespace Utilities.Security.Cryptography.X509Certificates
{
    public static class Crypto
    {
         /// <summary>
        /// Parses an RSA private key using the ASN.1 format.
        /// </summary>
        /// <param name="privateKey">Byte array containing PEM string of private key.</param>
        /// <returns>An instance of <see cref="RSACryptoServiceProvider"/> representing the requested private key.</returns>
        public static RSACryptoServiceProvider DecodeRsaPrivateKey(byte[] privateKey)
        {
            using (var memoryStream = new MemoryStream(privateKey))
            {
                using (var binaryReader = new BinaryReader(memoryStream))
                {
                    ushort shortValue = binaryReader.ReadUInt16();
                    switch (shortValue)
                    {
                        // If true, data is little endian since the proper logical seq is 0x30 0x81
                        case 0x8130: binaryReader.ReadByte(); break;
                        case 0x8230: binaryReader.ReadInt16(); break;
                        default: throw new CryptographicException("Improper ASN.1 format");
                    }

                    // (version number)
                    shortValue = binaryReader.ReadUInt16();
                    if (shortValue != 0x0102) throw new CryptographicException("Improper ASN.1 format, unexpected version number");

                    byte byteValue = binaryReader.ReadByte();
                    if (byteValue != 0x00) throw new CryptographicException("Improper ASN.1 format");

                    // The data following the version will be the ASN.1 data itself, which in our case
                    // are a sequence of integers.
                    // In order to solve a problem with instancing RSACryptoServiceProvider
                    // via default constructor on .net 4.0 this is a hack
                    var @params = new CspParameters
                    {
                        Flags = CspProviderFlags.NoFlags,
                        KeyContainerName = Guid.NewGuid().ToString().ToUpperInvariant(),
                        ProviderType = ((Environment.OSVersion.Version.Major > 5) || ((Environment.OSVersion.Version.Major == 5) && (Environment.OSVersion.Version.Minor >= 1))) ? 0x18 : 1
                    };

                    var rsaProvider = new RSACryptoServiceProvider(@params);
                    var rsaParams = new RSAParameters
                    {
                        Modulus = binaryReader.ReadBytes(DecodeIntegerSize(binaryReader))
                    };

                    // Argh, this is a pain.  From emperical testing it appears to be that RSAParameters doesn't like Byte buffers that
                    // have their leading zeros removed.  The RFC doesn't address this area that I can see, so it's hard to say that this
                    // is a bug, but it sure would be helpful if it allowed that. So, there's some extra code here that knows what the
                    // sizes of the various components are supposed to be.  Using these sizes we can ensure the buffer sizes are exactly
                    // what the RSAParameters expect.  Thanks, Microsoft.
                    var traits = new RSAParameterTraits(rsaParams.Modulus.Length * 8);
                    rsaParams.Modulus = rsaParams.Modulus.PadLeft(traits.SizeMod, 0x00);
                    rsaParams.Exponent = binaryReader.ReadBytes(DecodeIntegerSize(binaryReader)).PadLeft(traits.SizeExp, 0x00);
                    rsaParams.D = binaryReader.ReadBytes(DecodeIntegerSize(binaryReader)).PadLeft(traits.SizeD, 0x00);
                    rsaParams.P = binaryReader.ReadBytes(DecodeIntegerSize(binaryReader)).PadLeft(traits.SizeP, 0x00);
                    rsaParams.Q = binaryReader.ReadBytes(DecodeIntegerSize(binaryReader)).PadLeft(traits.SizeQ, 0x00);
[... 191 lines cut ...]
        /// <param name="value">The source byte array.</param>
        /// <param name="totalLength">The number of elements in the resulting array, equal to the number of original elements plus any additional padding elements.</param>
        /// <param name="element">A padding element.</param>
        /// <returns>The result byte array.</returns>
        public static byte[] PadLeft(this byte[] value, int totalLength, byte element)
        {
            //see https://stackoverflow.com/questions/25028609/how-do-i-left-pad-a-byte-array-efficiently
            if (value == null)
            {
                throw new ArgumentNullException("value");
            }

            if (value.Length >= totalLength)
            {
                return value;
            }

            var result = Enumerable.Repeat(element, totalLength).ToArray();
            long startIndex = totalLength - value.Length;
            Array.Copy(value, 0, result, startIndex, value.Length);
            return result;
        }
    }
}
```

josesimoes, 2019-07-18:
Closed with nanoFramework.System.Net#89.

closed 2019-07-18 by josesimoes