nanoframework/Home #508: nanoFramework.System.Net.Http.Client causes type resolution failure on ESP32

info-lvsys, 2019-07-28:
## Details about Problem

**nanoFramework area:** Nuget packages: nanoFramework.System.Net.Http.Client
(same error/problem with nanoFramework.System.Net.Http)

**VS version:** VS 2017 15.9.12

**VS extension version:** 1.2.03

**Target:** ESP32 DevKitV1

**Firmware image version:** 1.0.4.533

**Device capabilities output:**

System Information
HAL build info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Jul 26 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:
  HttpWebRequestSample, 1.0.0.0
  nanoFramework.Runtime.Native, 1.0.7.0
  nanoFramework.Runtime.Events, 1.0.8.0
  System.Net, 1.2.0.0
  mscorlib, 1.2.6.0
  System.Net.Http, 1.0.6.0
  nanoFramework.Networking.Sntp, 1.0.7.0

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.5.0, checksum 0x9ACFB207
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.2.0, checksum 0x7721D4BC
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

--------------------------------
::        Memory Map          ::
--------------------------------
  Type     Start       Size
--------------------------------
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

---------------------------------------------------------------
::                   Flash Sector Map                        ::
---------------------------------------------------------------
  Region     Start      Blocks   Bytes/Block    Usage
---------------------------------------------------------------
      0    0x00010000       1      0x100000     nanoCLR
      1    0x00110000       1      0x1C0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

---------------------------------------------------
::              Storage Usage Map                ::
---------------------------------------------------
  Start        Size (kB)           Usage
---------------------------------------------------
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x100000 (1024kB)   nanoCLR
  0x00110000    0x1C0000 (1792kB)   Deployment

Deployment Map
Empty

## Description

Referencing the nanoFramework.System.Net.Http.Client crashes the framework immediately when debugging the app.

It crashes with message: Exception starting CLR on device: Device stopped after type resolution failure. Can't start execution..

## Detailed repro steps so we can see the same problem

1. Load the sample HTTP project. Make zero code changes.

2. Update the nuget packages

3. Connect the ESP32 board.

4. Start the debug session

5. Crash with following error message: 

"Exception starting CLR on device: Device stopped after type resolution failure. Can't start execution..]"

## Expected behaviour

Application should start.

### Sample Project
Sample HTTP project from the sample folders.

## Screenshot
2019-07-27 18:37:00Z [Check and start debug engine on nanoDevice.]
2019-07-27 18:37:00Z [NanoDevices: open device \\?\USB#VID_10C4&PID_EA60#0001#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-07-27 18:37:00Z [Connect successful.]
2019-07-27 18:37:01Z [Erase deployment area successful.]
2019-07-27 18:37:02Z [Computing deployment blob.]
2019-07-27 18:37:04Z [Deploying assemblies.]
2019-07-27 18:37:05Z [Incrementally deploying assemblies to the device]
2019-07-27 18:37:06Z [Deployed 95612/95612 bytes.]
2019-07-27 18:37:06Z [Deployed assemblies with a total size of 95612 bytes.]
2019-07-27 18:37:06Z [Assemblies successfully deployed to the device.]
2019-07-27 18:37:07Z [Assigning debug engine of target device.]
2019-07-27 18:37:07Z [Attempting to update device debugger flags (1/5).]
2019-07-27 18:37:11Z [Device is running CLR, requesting reboot and pause for debugger (1/5).]
2019-07-27 18:37:13Z [Exception starting CLR on device: Device stopped after type resolution failure. Can't start execution..]
2019-07-27 18:37:13Z [NanoDevices: close device \\?\USB#VID_10C4&PID_EA60#0001#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]

josesimoes, 2019-07-28:
The message above is most likely because there is a managed app referencing a class lib that has a "wrong" version.
. NET assemblies have to have the correct refernenced assemblies available. That's what the type resolution step of the boot. And about what the device is complaining about. 

Make sure that you erase the deployment area and that you're referencing the correct Nugets. 

HttpRequest sample is probably outdated on the latest Nugets...

info-lvsys, 2019-07-29:
I cleaned everything out again. Restarted a small project from scratch. Same issue. 

You can reproduce this by:
- creating a blank project with solution file in same directory to ensure the packages folder stays local, 
- add the nuget package for nanoframework.System.Net.Http, 
- install all the dependencies.
- erase deployment on target
- start debug.
- fail. 

I've attached my sample project. Fresh from a minute ago.
[NFApp2.zip](https://github.com/nanoframework/Home/files/3439973/NFApp2.zip)

josesimoes, 2019-07-29:
This is indeed an issue with the assembly built against System.Net 1.2.0.

closed 2019-07-29 by josesimoes

GianniDPC, 2019-07-30:
This is not fixed? I get the same problem using nanoFramework.System.Net and nanoFramework.System.Net.Http. If I remove nanoFramework.System.Net.Http the application works but ofcourse I can't use classes like HttpClient then.

josesimoes, 2019-07-30:
This was fixed yesterday and the NuGet published.

Check for app.settings on your project and remove it. In doubt remove the NuGets and add them again.

info-lvsys, 2019-07-30:
How long does it take to update on Nuget? I am still getting the old one:

[image]

info-lvsys, 2019-07-30:
Nevermind, had to do it manually through the package manager console window:
PM> Install-Package nanoFramework.System.Net.Http -Version 1.0.6-preview.33 -Source https://www.myget.org/F/nanoframework-dev/api/v3/index.json

info-lvsys, 2019-07-31:
Thank you :) Works now

bmecdev, 2019-09-23:
Hi Im having the same issue whit the nanoFrameWork.Windows.Devices.Gpio'

******************************************************************************
** Error: Device stopped after type resolution failure.                     **
** Check in each assembly which assemblies are referenced and their version **
******************************************************************************
The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors.
The debugger engine thread has terminated unexpectedly with error 'Device stopped after type resolution failure. Can't start execution.'.

I'm using the latest image and upload to the latest nuget packages

System Information
HAL build info: 1.1.0.0, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Sep 16 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.1.0.0, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.6.0, checksum 0x9E949BB1
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.3.0, checksum 0x0B718489
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x110000     nanoCLR
      1    0x00120000       1      0x1B0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x110000 (1088kB)   nanoCLR
  0x00120000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty