nanoframework/Home #509: SslStream crashes MCU or throws SocketException in what appears to be random

info-lvsys, 2019-07-29:
Please read the following information before posting the issue.

**nanoFramework area:** C# code SslStream

**VS version:** VS 2017 15.9.12

**VS extension version:**1.2.0.3

**Target:** ESP32 WROOM
 
**Firmware image version:** 1.0.4.533

**Device capabilities output:**

```
System Information
HAL build info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Jul 26 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.5.0, checksum 0x9ACFB207
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.2.0, checksum 0x7721D4BC
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

--------------------------------
::        Memory Map          ::
--------------------------------
  Type     Start       Size
--------------------------------
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

---------------------------------------------------------------
::                   Flash Sector Map                        ::
---------------------------------------------------------------
  Region     Start      Blocks   Bytes/Block    Usage
---------------------------------------------------------------
      0    0x00010000       1      0x100000     nanoCLR
      1    0x00110000       1      0x1C0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

---------------------------------------------------
::              Storage Usage Map                ::
---------------------------------------------------
  Start        Size (kB)           Usage
---------------------------------------------------
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x100000 (1024kB)   nanoCLR
  0x00110000    0x1C0000 (1792kB)   Deployment

Deployment Map
Empty
```

## Description

The SslStream class is unexpectedly hanging, crashing MCU or throwing SocketException at various stages when executing HTTPS requests at regular interval (Example: every 5 minutes). Sometimes it crashes during the Socket.Close(), sometimes during the SslStream.AuthenticateAsClient().

## Detailed repro steps so we can see the same problem

1. Load the attached solution.
2. Configure the Wifi in Program.cs
3. Start the program and watch the console window for the output
4. The program will connect to wifi first
5. Then it will connect to howsmyssl.com (without validating cert) at regular intervals (30 to 60 seconds).

It should crash, eventually. 
...

## Expected behaviour

- Repeated calls to the same web server should work when there have been no changes to the web server SSL
- MCU should not crash on SslStream.Close() or Socket.Close()

### Sample Project
[NFApp2.zip](https://github.com/nanoframework/Home/files/3440717/NFApp2.zip)

## Screenshot
[image]

## Additional context

This was discovered when sending regular payloads to a web app on a 5 minute interval.

info-lvsys, 2019-07-29:
I am adding another sample project here, with more try..catch statements as was suggested on the ESP32 Discord channel.
[NFApp2.v2.zip](https://github.com/nanoframework/Home/files/3443273/NFApp2.v2.zip)

This morning, the attempts failed to close the SslStream (board hung, then reset). You can see the output from howsmyssl.com, then the close statement never finished.

```
Updating nanoDevice debugger engine.
The nanoDevice runtime is loading the application assemblies and starting execution.
'nanoFramework.Tools.VS2017.Extension.dll' (Managed): Loaded 'C:\Users\Max\Source\Repos\microcontrollers\NFApp2\bin\Debug\NFApp2.exe', Symbols loaded.
'nanoFramework.Tools.VS2017.Extension.dll' (Managed): Loaded 'C:\Users\Max\Source\Repos\microcontrollers\NFApp2\packages\nanoFramework.System.Net.1.2.0-preview.1\lib\System.Net.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'nanoFramework.Tools.VS2017.Extension.dll' (Managed): Loaded 'C:\Users\Max\Source\Repos\microcontrollers\NFApp2\packages\nanoFramework.Windows.Devices.Wifi.1.0.7-preview.17\lib\Windows.Devices.Wifi.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'nanoFramework.Tools.VS2017.Extension.dll' (Managed): Loaded 'C:\Users\Max\Source\Repos\microcontrollers\NFApp2\packages\nanoFramework.Networking.Sntp.1.0.7-preview.1\lib\nanoFramework.Networking.Sntp.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'nanoFramework.Tools.VS2017.Extension.dll' (Managed): Loaded 'C:\Users\Max\Source\Repos\microcontrollers\NFApp2\packages\nanoFramework.Runtime.Events.1.0.8-preview.20\lib\nanoFramework.Runtime.Events.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
Hello world!
*INFO* NetworkService.ctor() new instance. Requires DateTime = True
*INFO* NetworkService.WifiThread() started
*INFO* NetworkService.WifiThread() scanning
*INFO* NetworkService.Wifi_AvailableNetworksChanged() - get report
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :Home Brew,  BSSID : B4-75-0E-63-8B-E4,  rssi : -78,  signal : 2
*INFO* NetworkService.Wifi_AvailableNetworksChanged() Connected to Wifi network
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :xfinitywifi,  BSSID : F6-AA-A0-F7-07-D8,  rssi : -80,  signal : 2
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :GrandpaGlenn,  BSSID : EC-AA-A0-F7-07-D8,  rssi : -80,  signal : 2
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :Mikes,  BSSID : 74-7D-24-B5-03-18,  rssi : -85,  signal : 1
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :HOME-3A3F-2.4,  BSSID : D8-97-BA-F1-20-58,  rssi : -88,  signal : 1
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :Interweb,  BSSID : 20-4E-7F-9F-1E-12,  rssi : -91,  signal : 1
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :NETGEAR51,  BSSID : 84-1B-5E-E5-C0-89,  rssi : -91,  signal : 1
*INFO* NetworkService.Wifi_AvailableNetworksChanged() found WIFI: SSID :WU TANG LAN,  BSSID : 10-BE-F5-C3-9D-48,  rssi : -94,  signal : 1
*INFO* NetworkService.NetworkAvailabilityChanged() availability changed: IsAvailable = False
*INFO* NetworkService.NetworkAvailabilityChanged() availability changed: IsAvailable = True
*INFO* NetworkService.NetworkAddressChanged() address changed.
*INFO* NetworkService.CheckIP() Checking for IP
*INFO* NetworkService.CheckIP() got valid IP: 192.168.1.134
*INFO* NetworkService.NetworkAddressChanged() SNTP: setting up time
*INFO* NetworkService.SetDateTime() started checking
*INFO* NetworkService.SetDateTime() got valid date & time. UTC time is: 07/29/2019 16:18:11
Program.Main() time's up: checking URL
GetUrlSsl() Opening socket...
GetUrlSsl() Connecting...
GetUrlSsl() Authenticating with server...
GetUrlSsl() SSL handshake OK!
GetUrlSsl() Wrote 213 bytes
GetUrlSsl() Read 0 bytes, but only kept 6289
HTTP/1.0 200 OK
Accept-Ranges: bytes
Content-Length: 6080
Content-Type: text/html; charset=utf-8
Last-Modified: Thu, 16 May 2019 17:20:30 GMT
Vary: Accept-Encoding
Date: Mon, 29 Jul 2019 16:18:18 GMT

<!DOCTYPE html>
<html>
  <head>
    <meta charset="utf-8">
    <title>API &middot; How's My SSL?</title>
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta http-equiv="X-UA-Compatible" content="IE=edge" />
    
    <script>
     (function(i,s,o,g,r,a,m){i['GoogleAnalyticsObject']=r;i[r]=i[r]||function(){
         (i[r].q=i[r].q||[]).push(arguments)},i[r].l=1*new Date();a=s.createElement(o),
                              m=s.getElementsByTagName(o)[0];a.async=1;a.src=g;m.parentNode.insertBefore(a,m)
     })(window,document,'script','//www.google-analytics.com/analytics.js','ga');

     ga('create', 'UA-46659537-1', 'howsmyssl.com');
     ga('send', 'pageview');

    </script>

    <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/bootstrap/3.3.7/css/bootstrap.min.css" integrity="sha384-BVYiiSIFeK1dGmJRAkycuHAHRg32OmUcww7on3RYdg4Va+PmSTsz/K68vbdEjh4u" crossorigin="anonymous">

    <script
        s
GetUrlSsl() Closing ssl stream
The program '[1] nanoFramework application: Managed' has exited with code 0 (0x0).
```

josesimoes, 2019-07-30:
Confirmed issue.
Seems to be related with wrong handling of secure stream closing.

info-lvsys, 2019-08-01:
Hi @josesimoes - let me know when you want me to retry testing. I see you changed the status to "fixed" so I went ahead and tried on the latest  1.0.4.537 version, but the same project continues to crash for me, so I am not sure if it's in that release yet. Let me know, and I am more that happy to continue testing and thanks for having jumped on this ticket :)

josesimoes, 2019-08-01:
You're welcome!  Network TLS is a feature that sets nanoFramework apart so it's critical that it performs correctly. Thank you for spotting this and report the issue.

josesimoes, 2019-08-01:
The PR fixing this is being reviewed. As soon as it's merged you'll see this issue close. Then you can test it again.

closed 2019-08-04 by AdrianSoundy