nanoframework/Home #510: Assembly Deployment

haeberle, 2019-07-29:
The VS deploys all assembles when pressing start-debug also in case no code changes happened and the latest assembly version is already deployed. In case a project has several assemblies, this takes a long time to start the debugger.

Second, it looks like a multi referenced assemblies is deployed several times, with each lib referring to it.

I.e.
lib A is referenced in lib B & C, 
B&C are referenced in main, as well as lib A
A is deployed when B & C and main is deployed.

Here as well... really time consuming. 

**nanoFramework area:** Visual Studio extension

**VS version:** 2017

**VS extension version:** 15.9.8

**Target:** ESP32

**Firmware image version:**

**Device capabilities output:**

System Information
HAL build info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Jul 26 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.0.4.533, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:
  IotGatewayNano, 1.0.0.0
  mscorlib, 1.2.6.0
  nanoFramework.Json, 1.3.1.0
  System.Net, 1.2.0.0
  nanoFramework.Runtime.Events, 1.0.8.0
  Windows.Storage.Streams, 1.0.7.0
  nanoFramework.Runtime.Native, 1.0.7.0
  nanoFramework.Networking.Sntp, 1.0.7.0
  CommonLib, 1.0.0.0
  System.Math, 1.0.6.0
  ServiceBus, 1.0.0.0
  KNXService, 1.0.0.0
  KNXLib, 1.0.0.0
  Factories, 1.0.0.0
  MQTTService, 1.0.0.0
  PortalConnector, 1.0.0.0
  System.Net.Http, 1.0.6.0
  MSMqtt.nanoFramework, 4.3.0.0

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.5.0, checksum 0x9ACFB207
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.2.0, checksum 0x7721D4BC
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

--------------------------------
::        Memory Map          ::
--------------------------------
  Type     Start       Size
--------------------------------
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

---------------------------------------------------------------
::                   Flash Sector Map                        ::
---------------------------------------------------------------
  Region     Start      Blocks   Bytes/Block    Usage
---------------------------------------------------------------
      0    0x00010000       1      0x100000     nanoCLR
      1    0x00110000       1      0x1C0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

---------------------------------------------------
::              Storage Usage Map                ::
---------------------------------------------------
  Start        Size (kB)           Usage
---------------------------------------------------
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x100000 (1024kB)   nanoCLR
  0x00110000    0x1C0000 (1792kB)   Deployment

Deployment Map
Empty

## Description

A clear and concise description of what the bug is.

## Detailed repro steps so we can see the same problem

1. create a larger project with some libs 

2. press deploy or start debug

...

## Other suggested things

## Expected behaviour
Deploy only changed libraries and deploy a library only once

## Screenshot

Start debug
[image]
only deployment to the device
[image]

## Additional context

Add any other context about the problem here.

josesimoes, 2019-07-29:
The deployment step deploy all the referenced assemblies in the start project. That's by design. The alternative would be to download whatever is in the target device, compare with whatever is to the deploy and decide what to send over. Way more time consuming. 

Plus, because the deployment space is filled sequentially, if it happens that one of the changed assemblies was one in the middle that would require to deploy everything anyways... 

Also deploying requires erasing flash blocks in order to re-write them, so having to manage this and decide what to erase, not erase, push forward, etc will be rapidly become a nightmare! 

What you can do to improve this is to uncheck DEPLOY on all projects, except your main application (start project).

haeberle, 2019-07-30:
Overall I agree, but it depends on the size of the solution. In my example the solution contains 10 projects and each of them references i.e. some common libs, the deployment process needs, ~10sec per project. 

Your workaround is ok, but there are maybe options to optimize the full process:

-Bundle only assemblies which are not already bundled in a previous project in the solution
-In case the assembly contains a timestamp or at least calculate the checksum on the deployed assemblies and compare this with the assemblies which have to be deployed. 
 
-In case that will not work, a dialog in which the user can decide to deploy the solution or only start the debugger without deployment on his own risk that assemblies are outdated.

josesimoes, 2019-08-01:
The deployment and start debug process it's an incredibly harder and complex process.
As it is now is the result of a few iterations which lead to what was considered the optimal and possible.

Showing a dialog on deploy breaks VS expected workflow for deployment. And what you are describing on the last sentence is the "attach debugger" workflow. Which was ditched for nanoFramework by design as it was considered not reliable.

The deploy provider it's implemented over VS extensibility components. The code is [here](https://github.com/nanoframework/nf-Visual-Studio-extension/blob/develop/source/VisualStudio.Extension/DeployProvider/DeployProvider.cs) by all means if you come up with a better (and quick) approach do start a PR!

Deploying only your start project is the most possible efficient approach. The referenced projects are scattered, all referenced assemblies are colleted (no matter how deep they are) and then filtered. After this process the deployment bundle is built and deployed to the device.
On your screenshots above the deployment should take long because it's deploying each project over and over. After you select only the start project it's distilled to the absolute minimum. 

As a final note: do keep in mind that we are in the embedded world realm...

closed 2019-08-01 by josesimoes