nanoframework/Home #512: nanoFramework callback system.

planar74, 2019-08-01:
**nanoFramework area:** Other tools

There is no feature to create custom events from native to managed code.

josesimoes, 2019-08-02:
Elaborating on the above: I believe @planar74 is suggesting that "generic" events are added to nF. These would be free to use by OEMs, for example, to use in Interop code.

AdrianSoundy, 2019-08-04:
I think it would be a good idea to create a couple of event slots available to custom/oem extensions that are not going to become  a part of the general nanoFramework. Or for use by new code until a offical event slot is available.

planar74, 2019-08-04:
As follows from the source code, the task is not trivial. Introducing an additional mask for custom events will not be enough. It is necessary to do the recognition of events by the "manufacturer", for example, through the GUID, otherwise it is likely that several third-party libraries use the same bitmask for different events.
Will there be a desire to change the existing system of events?
When I asked a question, I did not assume that everything could be so complicated.

josesimoes, 2019-08-07:
The simplest way is to add a couple of events, complelty functional from the native code up to the C# events assembly. Then it's up to the OEM to code the handlers. I don't see that much of an issue that several third parties use them. The overlap is highly unlikely. If any it would be up to the parties to overcome the situation by either remapping them ot adding it on a special build. 

At this point I don't see the need for complelty rewriting the events layer to acomodate generic and blank events. @planar74 do you have any specific suggestion that you would like to offer regarding this?

planar74, 2019-08-07:
No, not have specific suggestion. If adding a couple of OEM events is not difficult, I will only be glad to see such functionality.

closed 2019-08-22 by josesimoes

josesimoes, 2019-08-23:
Sample code [here](https://github.com/nanoframework/Samples/tree/master/samples/NativeEvents).

Blog post with detailed explanations about it [here](https://jsimoesblog.wordpress.com/2019/08/23/posting-native-events-in-nanoframework/).

planar74, 2019-08-26:
Exelent! Thanks!