nanoframework/Home #514: Guid.ToString() changes byte order

haeberle, 2019-08-14:
## System.Guid.ToString() changes the order of the bytes.

**nanoFramework area:** lib-CoreLibrary/source/nanoFramework.CoreLibrary/System/Guid.cs

**VS version:**

**VS extension version:**

**Target:**

**Firmware image version:1.2.6-preview.16**

**Device capabilities output:**

## Description

The first three marked (--->>) for iterations count backward, the last two forward. The issue is, that in the first 3 loops the chars array and the bytes array are running contrary. Means the first bytes in chars[] are from the third byte in the bytes array, this results in a wrong byte order.
`

           public override string ToString()
           {
            // default format is dddddddd-dddd-dddd-dddd-dddddddddddd

            byte[] bytes = ToByteArray();

            char[] chars = new char[36];

            int i = -1, j;

    --->> for (j = 3; j >= 0; --j)
            {
                chars[++i] = HexToChar((bytes[j] & 0xF0) >> 4);
                chars[++i] = HexToChar((bytes[j] & 0x0F));
            }

            chars[++i] = '-';
    --->> for (j = 5; j >= 4; --j)
            {
                chars[++i] = HexToChar((bytes[j] & 0xF0) >> 4);
                chars[++i] = HexToChar((bytes[j] & 0x0F));
            }

            chars[++i] = '-';
    --->>for (j = 7; j >= 6; --j)
            {
                chars[++i] = HexToChar((bytes[j] & 0xF0) >> 4);
                chars[++i] = HexToChar((bytes[j] & 0x0F));
            }

            chars[++i] = '-';

   --->> for (j = 8; j <= 9; ++j)
            {
                chars[++i] = HexToChar((bytes[j] & 0xF0) >> 4);
                chars[++i] = HexToChar((bytes[j] & 0x0F));
            }

            chars[++i] = '-';
   --->> for (j = 10; j <= 15; ++j)
            {
                chars[++i] = HexToChar((bytes[j] & 0xF0) >> 4);
                chars[++i] = HexToChar((bytes[j] & 0x0F));
            }

            return new string(chars);
        }

          
`

haeberle, 2019-08-15:
Maybe it makes sense to add an additional constructor(string value)  and a Parse / TryParse method as well. I use the following to Parse:

`
         public static bool TryParseGuidWithDashes(String guidString, ref Guid result)
        {
            int startPos = 0;
            //int temp;
            //long templ;
            //int currentPos = 0;

            // check to see that it's the proper length
            if (guidString[0] == '{')
            {
                if (guidString.Length != 38 || guidString[37] != '}')
                {                   
                    return false;
                }
                startPos = 1;
            }
            else if (guidString[0] == '(')
            {
                if (guidString.Length != 38 || guidString[37] != ')')
                {
                    return false;
                }
                startPos = 1;
            }
            else if (guidString.Length != 36)
            {
                return false;
            }

            if (startPos > 0)
            {
                guidString = guidString.Substring(1, 36);
                startPos = 0;
            }

            if (guidString[8 + startPos] != '-' ||
                guidString[13 + startPos] != '-' ||
                guidString[18 + startPos] != '-' ||
                guidString[23 + startPos] != '-')
            {                
                return false;
            }

            var splits = guidString.Split('-');

            if (splits.Length != 5)
            {
                return false;
            }
            else
            {
                var res = new byte[16];

                ParsePhrase(splits[0], 4, 0, ref res);
                ParsePhrase(splits[1], 2, 4, ref res);
                ParsePhrase(splits[2], 2, 6, ref res);
                ParsePhrase(splits[3], 2, 8, ref res);
                ParsePhrase(splits[4], 6, 10, ref res);

                result = new Guid(res);
                return true;
            }
        }

        private static void ParsePhrase(string phrase, int length, int index, ref byte[] res)
        {
            for (int i = 0; i < length; i++)
            {
                var sub = phrase.Substring(i * 2, 2);
                var b = Convert.ToByte(sub, 16);
                res[i + index] = b;
            }
        }
`

josesimoes, 2019-08-23:
The .NET source has a more efficient way of dealing with this.
Of course we don't need all those options as we only print the standard format without open/close brackets. Check the source code [here](https://referencesource.microsoft.com/#mscorlib/system/guid.cs,1194).

josesimoes, 2019-08-23:
Regarding the parse, that's a great suggestion. We should have a new constructor that accepts a string.
To simplify the accepted formats would be:

* {dddddddd-dddd-dddd-dddd-dddddddddddd}
* dddddddd-dddd-dddd-dddd-dddddddddddd

The .NET source is [here](https://referencesource.microsoft.com/#mscorlib/system/guid.cs,673). Just for comparison.

Roadrunner67, 2019-09-25:
I have a TryParseGuidWithDashes ready for PR, but please explain what you think is wrong with the current ToString(). If I create a .Net Guid from "a8b287c6-357f-46a8-9b9c-152daaeaa003" it converts to a bytearray containing:  198 135 178 168 127 53 168 70 155 156 21 45 170 234 160 3.
A nanoFramewor Guid created with this byte array resolves with ToString() to: "a8b287c6-357f-46a8-9b9c-152daaeaa003". I dont see the problem!?

Roadrunner67, 2019-09-26:
I have made a [PR](https://github.com/nanoframework/lib-CoreLibrary/pull/88) which introduces the string constructor.
Apart from that we're waiting for @haeberle  to elaborate on the nature of the problem he is facing with Guid.ToString().

haeberle, 2019-11-08:
The initial issue "System.Guid.ToString() changes the order of the bytes" is not fixed

haeberle, 2019-11-08:
pls set on my reopen on hold.... further investigation needed

josesimoes, 2019-11-11:
@haeberle please clarify: is this still an issue or not?

closed 2019-11-13 by josesimoes