nanoframework/Home #517: Add support for extended RAM

AdrianSoundy, 2019-08-20:
Support the Extended RAM if available.

The IDF libraries need to be updated to enable the PSRAM option. This mainly affects the Bootloader. 

Need to test if we can have just one build with PSRAM enabled or do we need a new option.  
Will this build also work on a non PSRAM ESP32.  There are indications it will in the IDF driver code.

Existing build already has support for allocating the managed heap in PSRAM if it is available(Tested). May need some adjustment. See nanoClr\memory.cpp

This issue is just part of updating the IDF to latest version.

AdrianSoundy, 2020-02-01:
This has been completed as per using IDF 3.3 but an issue has arisen when using 64bit PSRAM as not recognised by the bootloader.

Requires config option for PSRAM to be set to Auto Detect instead of 32 bit.
This can be done as part of the IDF 3.3.1 update

josesimoes, 2020-02-21:
@AdrianSoundy IIRC this was brought in with the latest move to IDF 3.3.1.
I guess this can this be closed now...

AdrianSoundy, 2020-02-27:
Yes it can be closed

closed 2020-02-27 by AdrianSoundy