nanoframework/Home #518: Add support for the ESP32 Remote Control (RMT) peripheral

marknotgeorge, 2019-08-21:
The ESP32 chip has a Remote Control (RMT) peripheral, which facilitates transmitting and receiving infra-red remote control signals. Further details can be found in the [IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html).

It would also be useful to add support for IR code formats and protocols. Support for code databases such as [irdb](http://irdb.tk/) might be an idea.

josesimoes, 2019-08-23:
Nice that you are looking into this! 👏 

Being a "niche" feature the use scenarios are very specific. Meaning that the inclusion of the native code should be dependent of a build feature. For the same reason the supporting API should be a class library on its own.

marknotgeorge, 2019-08-23:
@ololoshka2871 has already been [working on this](https://github.com/ololoshka2871/nanoFramework.Hardware.Esp32.RMT)

josesimoes, 2019-12-01:
Still active, correct @marknotgeorge 😁?

closed 2020-07-06 by josesimoes