nanoframework/Home #520: HttpListener.GetContext() fails when listening on HTTPS (throws SocketException)

info-lvsys, 2019-08-23:
## Details about Problem

**nanoFramework area:** Nuget packages: System.Net.Http

**VS version:** VS2017

**VS extension version:** 1.2.3.24

**Target:** ESP32_WROOM_32

**Firmware image version:** 1.0.5.0

**Device capabilities output:**

System Information
HAL build info: 1.0.5.0, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Aug  7 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.0.5.0, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.5.0, checksum 0x9ACFB207
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.2.0, checksum 0x7721D4BC
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

--------------------------------
::        Memory Map          ::
--------------------------------
  Type     Start       Size
--------------------------------
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

---------------------------------------------------------------
::                   Flash Sector Map                        ::
---------------------------------------------------------------
  Region     Start      Blocks   Bytes/Block    Usage
---------------------------------------------------------------
      0    0x00010000       1      0x110000     nanoCLR
      1    0x00120000       1      0x1B0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

---------------------------------------------------
::              Storage Usage Map                ::
---------------------------------------------------
  Start        Size (kB)           Usage
---------------------------------------------------
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x110000 (1088kB)   nanoCLR
  0x00120000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

## Description

HttpListener fails to accept connections over SSL. When a connection is accepted, a SocketException is immediately raised, before the browser can offer to trust the SSL certificate.

## Detailed repro steps so we can see the same problem

1. Run the attached sample project.

2. Take note of the IP address your device is listening on (IP_ADDRESS) displayed in visual studio output

3. Open a web browser to https://IP_ADDRESS

4. Watch the visual studio output for the SocketException.

4. Web page should be broken.

## Expected behaviour

A successful connection: web page should show "It Works"

You can try changing the prefix to "http" and repeat above procedure with http://IP_ADDRESS Should display "it works" in the browser 

### Sample Project
[NFApp3.zip](https://github.com/nanoframework/Home/files/3532675/NFApp3.zip)
Change the SSID and PASSWORD to yours before running it.

## Screenshot
[image]

info-lvsys, 2019-08-23:
This is a bug also present in the SslStream.AuthenticateAsServer(). I did a lot of research into it. I am wondering if the SecureServerInit works properly. The 509 certificate passed must include the private key, and be a 509Certificate2() 

Attached is a .NET Core equivalent. Has to use a PFX with a password (see comments in the code):

[DotNetCoreApp.zip](https://github.com/nanoframework/Home/files/3532922/DotNetCoreApp.zip)

josesimoes, 2019-11-12:
Confirmed issue in `AuthenticateAsServer` methods and also in native implementation of SSL accept code.

closed 2019-11-13 by josesimoes

josesimoes, 2019-11-13:
@info-lvsys the thing with X502 certificates shouldn't be a problem anymore. There was indeed a bug in the `AuthenticateAsServer`.

info-lvsys, 2019-11-13:
@josesimoes awesome! thank you for fixing it. Had other priorities for a while but will resume testing asap.