nanoframework/Home #526: Communication breaking between Visual Studio and ST-LINK (uart)

valoni, 2019-09-09:
## Details about Problem

**Vs:** VS2017 (1.2.4.8 VSIX)

**Target:** Community-Firmware (Nucle STM32F411RET6)

**Firmware image version:** 1.1.186.0

Seems after updated version of Visual Studio Debuger 1.2.3.x break communication between Visual Studio and ST-LINK Communication 

## Description

Randomly recognise and again dissaper COM from Device Explorer even if i restart device or unplug from power 

## Detailed repro steps so we can see the same problem

1.try to open Device Explorer to see COM from Nucleo STM32F411RET5 if you see it try to deploy and restart Hardware-device

...

## Other suggested things

## Expected behaviour

COM should appear always on Device Explorer if COM port not in use

## Screenshot
[image]

[image]

so after i've got 
2019-09-08 09:52:19Z [NanoDevices: checking device \?\USB#VID_0483&PID_374B&MI_02#6&4a8369c&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-09-08 09:52:21Z [NanoDevices: quitting device \?\USB#VID_0483&PID_374B&MI_02#6&4a8369c&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-09-08 09:52:21Z [NanoDevices: Serial device enumeration completed. Found 0 devices]
2019-09-08 09:52:34Z [NanoDevices: checking device \?\USB#VID_0483&PID_374B&MI_02#6&4a8369c&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-09-08 09:52:36Z [NanoDevices: quitting device \?\USB#VID_0483&PID_374B&MI_02#6&4a8369c&0&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2019-09-08 09:52:36Z [NanoDevices: Serial device enumeration completed. Found 0 devices]

jabacz, 2019-09-09:
Hi, I'm the one who got the issue and asked about @Discord. Few additional information.
- the target acts the same with 183, 184, 185, 186
- after flashing the NUCLEO, in my case STM32F411RE6, it appears in device explorer
- any attempt to communicate eg. PING, List capabilities, Depoy ends with "no response"
- since than the device cannot be found by VS anymore, reconnecting, resetting does not help
- if target is re-flashed, than appears to be discovered, but any connect breaks it again
- nF extension 1.2.3.4 in my case
- ST_Link and drivers in version downloaded on Friday Sep 6th
- the device in "broken state" works normally with ST_Link and Arduino studio
- I have tried a fresh new installation of Win10 1903 with up-to-date updates in VmWare with USB redirected to the VM. The behavior is exactly the same. Eg. VS discovers the device, any attempt to communicate breaks it until re-flashed.

josesimoes, 2019-09-13:
This is seems to have improved after VS extension v1.2.5.31 (VS2017).

valoni, 2019-09-14:
[image]

i can confirm everything is working well with latest extension v1.2.5.31 (can deploy,ping ,erase and show device capabilities)

closed 2019-09-14 by valoni